tomquist/esphome-b2500 #1: [ble_sensor:078] Error reading char at handle 21, status=2 => Keine Bluetooth Verbindung wird aufgebaut

RALFISBORN, 2024-06-18:
Hallo,

ich habe folgendes Setup:
+ B2500 Greensolar V2, FW 163
+ ESP32: az-delivery-devkit-v4

Wifi Verbindung und MQTT Verbindung steht.

Allerdings wird die Bluetooth Verbindung nicht aufgebaut:
[image]

So ist in der YAML - File das Board deklariert:
[image]

An was könnte das liegen?

Vielen Dank

tomquist, 2024-06-18:
Die Warnung kannst du ignorieren. Ansonsten sehen die Logs doch gut aus. Funktioniert irgendetwas nicht?

RALFISBORN, 2024-06-18:
Und diese Meldungen stehen noch im MQTT Log:
[image]

tomquist, 2024-06-18:
Woher kommen die Logs? Die kommen nicht vom ESP32. Kann es sein, dass etwas beim Einrichten der Topics in deiner Heimautomatisierung nicht stimmt? Was Zeigt die Weboberfläche des ESP32 an? Sind da Werte zu sehen?

tomquist, 2024-06-18:
Zu beachten ist, dass der Speicher nicht mit der App per Bluetooth verbunden sein darf. Ansonsten kann sich der ESP32 nicht verbinden da der Speicher nur eine gleichzeitige Bluetooth Verbindung unterstützt.

RALFISBORN, 2024-06-18:
> Woher kommen die Logs? Die kommen nicht vom ESP32. Kann es sein, dass etwas beim Einrichten der Topics in deiner Heimautomatisierung nicht stimmt? Was Zeigt die Weboberfläche des ESP32 an? Sind da Werte zu sehen?

Das kommt von IOBroker.
Da sind die Einstellungen wie folgt:
[image]

[image]

Was mich z.B. wundert, ist die MQTT Verbindung. Diese besteht definitiv, da ja auch Topics gepublisht werden und z.B. die Laufzeit mit der von der ESP Weboberfläche übereinstimmt. Trotzdem wird sie hier als nicht verbunden angezeigt:
[image]

Wenn ich z.B. den MQTT Broker neu starte, dann wird dies auch in den Logs des ESP angezeigt.

RALFISBORN, 2024-06-18:
> Zu beachten ist, dass der Speicher nicht mit der App per Bluetooth verbunden sein darf. Ansonsten kann sich der ESP32 nicht verbinden da der Speicher nur eine gleichzeitige Bluetooth Verbindung unterstützt.

Das ist mir auch aufgefallen. Wenn der ESP läuft, kann ich via App auch keine Verbindung via Bluetooth aufbauen. Sobald der ESP ausgesteckt ist, klappt es wieder. Sieht für mich so aus, als würde generell eine Verbindung bestehen.

Nur warum keine Werte ausgelesen werden, das wundert mich noch :(

tomquist, 2024-06-18:
Ich kenne mich mit HomeAssistant nicht so aus aber ich würde mal schauen, dass die MQTT Konfiguration in HomeAssistant passt. In deinen Logs sehe ich jede Menge dieser Meldungen:
> ... Received unknown status payload: ...

Das deutet darauf hin, dass HomeAssistant irgendwelche Nachrichten auf diese Topics publiziert, mit denen der ESP32 nichts anfangen kann. Ich vermute das kommt von der Setting "Eigene State beim Verbinden publizieren", die deaktiviert werden sollte. Außerdem sollte vermutlich "Unterschiedliche Topic Namen für setzen und lesen nutzen" aktiviert sein. Ich kann mir gut vorstellen, dass das alles negativen Einfluss auf die Stabilität des ESP32 hat.

RALFISBORN, 2024-06-19:
> Ich kenne mich mit HomeAssistant nicht so aus aber ich würde mal schauen, dass die MQTT Konfiguration in HomeAssistant passt. In deinen Logs sehe ich jede Menge dieser Meldungen:
>
>
> Das deutet darauf [...]

Ich nutze auch kein HomeAssistant. Die Screenshots sind von IOBroker. Das macht es auch schwierig für mich, da die Scripts und der Support wohl sehr auf HA ausgelegt sind.

Die MQTT Einstellungen habe ich entsprechend angepasst:
[image]

Leider ohne positives Ergebnis.
Nutzt du auch den IOBroker oder HA?

tomquist, 2024-06-19:
Ah, sorry. Ich meinte ioBroker. Kenne mich weder mit HomeAssistant, noch mit ioBroker aus da ich OpenHAB nutze. Eventuell hilft dir die Topic Liste hier auch für das Einrichten im ioBroker: https://github.com/noone2k/hm2500pub/wiki/ESP32-MQTT-TOPICS

Ansonsten kann weiß ich aktuell nicht was schief geht. Kannst du ml noch ein paar mehr Logs in text form hier posten?

RALFISBORN, 2024-06-19:
> Ah, sorry. Ich meinte ioBroker. Kenne mich weder mit HomeAssistant, noch mit ioBroker aus da ich OpenHAB nutze. Eventuell hilft dir die Topic Liste hier auch für das Einrichten im ioBroker: [...]

Das ist natürlich Schade, dass du hier nicht von der IOBroker Ecke bist.

Anbei mal das Log in Textform:

> 14:26:25 | [I] | [mqtt:274] | MQTT Connected!
> -- | -- | -- | --
> 14:26:25 | [W] | [component:237] | Component mqtt took a long time for an operation (130 ms).
> 14:26:25 | [W] | [component:238] | Components should block for at most 30 ms.
> 14:26:26 | [W] | [mqtt.button:023] | 'Restart Controller': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:26 | [W] | [mqtt.switch:032] | 'B2500: Bluetooth Enabled': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:26 | [W] | [mqtt.button:023] | 'B2500: Reboot B2500': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:26 | [W] | [mqtt.button:023] | 'B2500: Reset to Factory Settings': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:26 | [W] | [mqtt.number:022] | Can't convert '' to number!
> 14:26:26 | [W] | [mqtt.number:022] | Can't convert '' to number!
> 14:26:26 | [W] | [mqtt.switch:032] | 'B2500: Timer 1 Enabled': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:26 | [W] | [mqtt.switch:032] | 'B2500: Timer 2 Enabled': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:26 | [W] | [mqtt.switch:032] | 'B2500: Timer 3 Enabled': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:26 | [W] | [mqtt.switch:032] | 'B2500: LoadFirst': Received unknown status payload:
> 14:26:26 | [W] | [component:157] | Component mqtt set Warning flag: unspecified
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:26:31 | [W] | [component:170] | Component mqtt cleared Warning flag
> 14:27:09 | [W] | [ble_sensor:078] | Error reading char at handle 21, status=2
> 14:28:09 | [W] | [ble_sensor:078] | Error reading char at handle 21, status=2
> 14:28:23 | [I] | [ota:117] | Boot seems successful, resetting boot loop counter.
> 14:29:09 | [W] | [ble_sensor:078] | Error reading char at handle 21, status=2
> 14:32:09 | [W] | [ble_sensor:078] | Error reading char at handle 21, status=2

Der Tipp mit den MQTT Topics ist auf jeden Fall interessant. Denn dabei ist mir folgendes aufgefallen (hab mir bis jetzt nur mal den Battery Ordner angeschaut):
So sind z.B. die erzeugten Topics im MQTT Broker angelegt worden:
[image]

Mal abgesehen davon, dass hier die gelb markierten fehlen:
[image]

Zudem wird dann noch der Typ von z.B. DOD Set als String und nicht als Number angelegt:
[image]

Ist der Aufbau der Topics in der YAML Datei hinterlegt? Diese kann ja nicht manuell angepasst werden vor dem Flashen oder?

RALFISBORN, 2024-06-19:
Ich habe hier auch mal die Log Datei des MQTT Brokers vom IOBroker angehängt. 
Sieht für mich so aus, als würden die Topics nicht so angelegt werden, wie es der Broker gerne wünscht. Oder was meinst du?
[LogLevel_Alles.txt](https://github.com/user-attachments/files/15903201/LogLevel_Alles.txt)

tomquist, 2024-06-19:
Ich meinte eher Logs vom ESP32. Ich kann mit den ioBroker logs wenig anfangen und kann dir auch nicht wirklich helfen das im ioBroker einzurichten aber ich kann schauen ob ESP32-seitig alles passt.

RALFISBORN, 2024-06-19:
> Am Besten auch mal hier die Config posten, mit der du das Image baust. Die kannst du bekommen indem du auf den "Report an Issue" button drückst und dann den Teil ab `**Configuration**` hier rein kopierst.

Das wäre folgende hier: Habe nur die Wifi SSID und den MQTT Benutzer angepasst.
```
json
{
  "name": "b2500",
  "friendly_name": "b2500",
  "mqtt": {
    "topic": "b2500",
    "broker": "IP",
    "port": 1883,
    "username": "Benutzer",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "WIFI SSID",
    "password": "***"
  },
  "board": "az-delivery-devkit-v4",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": true,
  "enable_cmd30": false,
  "enable_esp_temperature": true,
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": true,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "ota": false,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "storages": [
    {
      "name": "b2500",
      "version": 2,
      "mac_address": "***"
    }
  ],
  "poll_interval_seconds": 5
}
```

RALFISBORN, 2024-06-19:
> Ich meinte eher Logs vom ESP32. Ich kann mit den ioBroker logs wenig anfangen und kann dir auch nicht wirklich helfen das im ioBroker einzurichten aber ich kann schauen ob ESP32-seitig alles passt.

Ich wüsste nicht was ich sonst an Logs herausholen könnte, außer das was ich oben bereits gepostet habe :(

RALFISBORN, 2024-06-19:
Hast du das mit dem Build Problem bzgl. alter Node.js Versionen oben gesehen? Liegt hier vielleicht auch das Problem mit der Erstellung der Firmware?

Kann die YAML auch nachträglich noch angepasst werden in deinem Tool, bevor daraus die Firmware erstellt wird?

tomquist, 2024-06-19:
Der Build schlägt fehl seit dem das Temperature feature eingeschaltet wurde. Das scheint mit deinem ESP32 board nicht kompatibel zu sein. Versuch das mal auszuschalten.

RALFISBORN, 2024-06-19:
> Der Build schlägt fehl seit dem das Temperature feature eingeschaltet wurde. Das scheint mit deinem ESP32 board nicht kompatibel zu sein. Versuch das mal auszuschalten.

Danke für den Tipp. Ja, an dem lags. Hab auch nochmal den Build erstellt aber ohne Erfolg. Gleiches Verhalten. Hätte ja sein können.

RALFISBORN, 2024-06-21:
Kann es eigentlich sein, dass sich mittlerweile kein Image mehr erstellen lässt? Zumindest wird der Vorgang immer mit Exit Code 1 abgebrochen.

tomquist, 2024-06-29:
@RALFISBORN Ich habe gerade einen bug bei der Initialisierung der Verbindung mit v2 behoben. Kannst du es nochmal mit der aktuellen Version versuchen?

RALFISBORN, 2024-06-30:
> @RALFISBORN Ich habe gerade einen bug bei der Initialisierung der Verbindung mit v2 behoben. Kannst du es nochmal mit der aktuellen Version versuchen?

habe es mittlerweile mit einem alternativen Code hinbekommen. Aber Danke ;)

closed 2024-06-30 by RALFISBORN