tomquist/esphome-b2500 #2: Build image not successfull

Rookie1967, 2024-06-21:
**Generating an ESPHome configuration to connect an ESP32 to a B2500 memory with the ESPHome B2500 Config Generator is not successful if the board "4d-systems-esp32s3-gen4-n8n16" is selected.**

**ESPHome B2500 Config Generator doesn't work if the board " 4d_systems_esp32s3_gen4_r8n16" was selected. When I select the board "esp32-s3-devkitc-1" the image is created. **

**Reproducible by selecting the board 4d_systems_esp32s3_gen4_r8n16"**

**build an image**

**Error when building the image**

**Screenshots**

**Device information**

- ESP Board: ESP32-S3-DevKitC-1 Board, APKLVSR WROOM-1-N16R8

- B2500 Brand: Greensolar

- B2500 Version: v3

**Configuration**

```json
{
  "name": "batterie",
  "friendly_name": "Batterie",
  "poll_interval_seconds": 5,
  "mqtt": {
    "topic": "batterie",
    "broker": "192.168.178.35",
    "port": 1883,
    "username": "mqtt",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "FRITZ!Box 7590 IN",
    "password": "***"
  },
  "board": "4d_systems_esp32s3_gen4_r8n16",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": false,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": false,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "storages": [
    {
      "name": "B2500",
      "version": 1,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2024-06-22:
The Board `4d_systems_esp32s3_gen4_r8n16` is unknown by ESPHome so it can't automatically detect the board variant. I added a new config option to explicitly set the variant. Can you try again but change "Variant" to "esp32s3"?

tomquist, 2024-06-22:
Ok, just tried this but it's still not working. I'm not sure how to solve this. Looks like this is a fairly new board so it just might not yet be supported by ESPHome. Did you try building the image in HomeAssistant?

Rookie1967, 2024-06-22:
Hi,

Thank you for quick support.

I tryed it.

Board: 4d_systems_esp32s3_gen4_r8n16 (=4D Systems GEN4-ESP32 16MB (ESP32S3-R8N16)) + Variant: esp32-s3---> build failed

Board: Espressif ESP32 Dev Module (esp32dev) + Variant: esp32-s3 ---> build failed

Board: Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM) (esp32-s3-devkitc-1) + Variant: esp32-s3 --> build successful b2500-esp32s3.factory.bin and b2500-esp32s3.ota.bin were built

I'm not sure which one I'd to install, so I' tested both.

I installed it in ESPHome.

factory.bin didn't work. I can't see the board in my network.

The same with ota.bin.

Can you recomment a ESP-board, which I can us for th B2500?

Best Regards

 

 

 

 

Gesendet: Samstag, 22. Juni 2024 um 11:46 Uhr

Rookie1967, 2024-06-22:
No, I didn' try building the image in HomeAssistant.

 
 

Gesendet: Samstag, 22. Juni 2024 um 12:09 Uhr

tomquist, 2024-07-06:
I was able to successfully flash the same board by selecting board `esp32-s3-devkitc-1` and variant `esp32s3`

Rookie1967, 2024-07-07:
Hallo,
Ich sehe grade, dass du auch in Deutschland wohnst und versuchs deshalb mal in deutsch, damit es durch Google-Translate bei mir nicht zu Missverständnissen kommt.
Irgendwie weigert sich mein ESP32 weiterhin, das Image zu übernehmen und damit zu laufen.
Da ich noch recht unerfahren in der IoT-Welt bin, schreibe ich mal in Stichworten, was ich mache. Wahrscheinlich ist da irgendwo der Fehler.
Der ESP wurde von mir  über web.esphome.io eingerichtet (Prepare device for first use) und WLAN ebenfalls eingerichtet.
Der ESP32 ist sowohl im WLAN als auch über Bluetooth zu finden.
Dann bin ich wie folgt vorgegangen:
1. Verbinden des ESP mit Windows-PC (linke USB-C-Buchse am ESP)
2. ESP32 über Netzwerkadresse augerufen
3. Image-Datei (b2500-esp32s3.factory.bin) über "Update" aufgespielt, was mit "erfolgreich" bestätigt wurde.
4. Meine "esphome-web-c85e94.yaml" vom angeschlossenen ESP sieht aber so aus wie vorher:

substitutions:
  substitutions:
  name: esphome-web-c85e94
  friendly_name: ESPHome Web c85e94

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32s3.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

7. Wenn ich über web.esphome.io das Image über "Install" aufspiele, wird mir ebenfalls "Configuration installed!" angezeigt.
8. Nach Reset vom ESP Bootet dieser immer wieder neu, ist aber weder im Netzwerk noch bei Bluetooth zu finden.
9. Im web.esphome.io wird dieses im Log:

[01:54:51]ESP-ROM:esp32s3-20210327
[01:54:51]Build:Mar 27 2021
[01:54:51]rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[01:54:51]SPIWP:0xee
[01:54:51]mode:QIO, clock div:1
[01:54:51]load:0x3fce3808,len:0x16c4
[01:54:51]ets_loader.c 78 
[01:54:51]ESP-ROM:esp32s3-20210327
[01:54:51]Build:Mar 27 2021
[01:54:51]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[01:54:51]Saved PC:0x400454d5
[01:54:51]SPIWP:0xee
[01:54:51]mode:QIO, clock div:1
[01:54:51]load:0x3fce3808,len:0x16c4
[01:54:51]ets_loader.c 78 
[01:54:51]ESP-ROM:esp32s3-20210327
[01:54:51]Build:Mar 27 2021
ab hier wiederholt es sich ständig.

10. Bei Home-Assistanit steht das:

INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/esphome-web-c85e94.yaml...
INFO Starting log output from esphome-web-c85e94.local using esphome API
WARNING Can't connect to ESPHome API for esphome-web-c85e94.local: Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)
INFO Trying to connect to esphome-web-c85e94.local in the background

Was mache ich falsch?

Ich würde auch gerne ein anderes ESP32-Board ausprobieren. Hast du da eine Empfehlung?

Das war's jetzt erstmal von hier.

Viele Grüße.

tomquist, 2024-07-07:
Ich habe gerade noch eine Option gepushed, die du versuchen kannst. Kannst du das ganze wie folgt konfigurieren:
[image]

und dann unter "Advanced" das hier aktivieren:
[image]

Rookie1967, 2024-07-07:
Hallo,
ES HAT FUNKTIONIERT !!! :-D

Ich bin begeistert und bewundere, was in so manchen klugen Köpfen drin steckt. Für mich sind das größtenteils alles böhmische Dörfer...
Vielen Dank für deine Hilfe. Was kann ich dir dafür Gutes tun?

Ich kann den esp32 jetzt über die ID in meinem Browser aufrufen und sehe die schicke Grafik mit den Werten und Einstellmöglichkeiten.

Im Debug Log gibt es noch zwei Warnungen, die sich aber nicht dramatisch anhören:
11:54:44	[W]	[component:237]	Component esp32_ble took a long time for an operation (80 ms).
11:54:44	[W]	[component:238]	Components should block for at most 30 ms.
Die wiederholen sich alle 10 Sekunden.

Das ESP sendet die Daten auch zu meinen mqtt-broker und ich sehe sie auch im mqqt explorer vom Home Assistant.

Bei ESPHome im Home Assistant kann ich das ESP32 aber nicht sehen, was mich in meiner Unwissenheit doch wundert.

Auf jeden Fall bin ich heute einen riesen Schritt weitergekommen in Richtung optimaler Solarenergienutzung. 

Die MQTT-Datenverbindung zum Hame-Server zu kappen und alles nur auf meinen mqqt broker zu schicken hatte ich mich bis jetzt nicht getraut. Dank deiner Hilfe komme ich nun an die Daten meiner Batterie und hab trotzdem noch eine Verbindung zu Hame. Bin halt eher konservativ eingestellt.

tomquist, 2024-07-07:
Dass HomeAssistant das Gerät nicht sieht, liegt vermutlich daran, dass bei dir der Haken unter MQTT bei "Discovery" nicht gesetzt ist:
[image]

Rookie1967, 2024-07-10:
Hallo, genau das ist es gewesen. 
Ich danke dir für die viele Hilfe, die du mir gegeben hast.

closed 2024-07-10 by tomquist

qwikser, 2024-09-11:
Hallo - beim Image erstellen bekomme ich immer diese Meldung ([build_esphome](https://github.com/tomquist/esphome-b2500/actions/runs/10817527800/job/30011105886#step:1:24)
This request has been automatically failed because it uses a deprecated version of `actions/upload-artifact: v2`. Learn more: https://github.blog/changelog/2024-02-13-deprecation-notice-v1-and-v2-of-the-artifact-actions/)

..hatte es vor 2 Tagen zum Test mit einem Test Wifi & MQTT versucht, dass muss ich aber jetzt ändern. Oder kann ich die CFG auch ohne neues Build anpassen irgendwo? :)

MfG.
Daniel