tomquist/esphome-b2500 #3: Keine Empfang einer IP

Black7Nomad, 2024-06-25:
Hallo,

die Installation läuft vollständig durch , allerdings erhalte ich immer die Fehlermeldung:
 
Couldn't resolve IP address for 'core-mosquitto'!

[image]

Habe ich hier irgendwo einen Fehler

 gemacht? Habe alles mögliche versucht, komme aber nicht weiter.

**Device information**

- ESP Board: ESP32

- B2500 Brand: Marstek

- B2500 Version: v2

**Configuration**

```json
{
  "name": "b2500",
  "friendly_name": "B2500",
  "poll_interval_seconds": 5,
  "mqtt": {
    "topic": "b2500",
    "broker": "core-mosquitto",
    "port": 1883,
    "username": "MQTT-USER",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "Speedport Rietenau",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "esp32",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": false,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": false
  },
  "storages": [
    {
      "name": "B2500",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2024-06-26:
Von den Logs her vermute ich mal dass der MQTT username und/oder das Passwort falsch sind

Black7Nomad, 2024-06-26:
Danke für die Hilfe, die Logs habe ich mehrfach kontrolliert, Nachdem ich den MQTT Broker und den benutzer komplett entfernt und neu aufgesetzt habe funktionierte es nun :-) 

Leider habe ich gleich das nächste Problem, vielleicht kannst du mir hier auch weiterhelfen...
wie du merkst bin ich nicht ganz der fitteste in dem Thema.

Nachdem der MQTT und die Bluetooth Verbindung aufgebaut ist erhalte ich nun diese Fehlereintrag im Log.

[W][ble_sensor:078]: Error reading char at handle 27, status=2

[image]

tomquist, 2024-06-26:
Soweit ich weiß kann man diese Warnung ignorieren. Funktioniert denn sonst alles andere? Startet der ESP danach neu? Falls ja, kannst du mal versuchen den Auto-Restart auszuschalten indem du unter "Advanced" die Checkbox "Enable Auto Restart" deaktivierst.

Black7Nomad, 2024-06-26:
Nein, der error counter geht bis auf 8 oder 9, dann erfolgt ein Neustart des Zählers. 
Ich bekomme im Home Assistant keine Werte angezeigt.
Also keine Entitäten etc. angelegt.
Grüße

tomquist, 2024-06-29:
@Black7Nomad Ich habe gerade einen bug bei der Initialisierung der Verbindung mit v2 behoben. Kannst du es nochmal mit der aktuellen Version versuchen?

Black7Nomad, 2024-06-29:
@tomquist 
Werde ich gleich morgen versuchen.
Ich danke dir. Sobald ich mehr weiß gebe ich Bescheid.

Black7Nomad, 2024-07-01:
@tomquist so konnte die neue Datei aufspielen, so sieht der Log nun aus.

[image]

tomquist, 2024-07-01:
Die Logs sehen gut aus. Scheint nun Das Script nutzt aktuell hauptsächlich MQTT. Entweder du legst die entities selbst in HomeAssistant an oder nutzt die Config von @neromatrix, die eine bessere HomeAssistant integration hat: https://gist.github.com/neromatrix/6ec812f35418c4b38ebbad4e92cac888

Eventuell verbessere ich irgendwann mal die HomeAssistant integration. Da ich selbst kein HomeAssistant nutze, ist die Prio dafür aber ziemlich niedrig.

closed 2024-07-01 by tomquist