tomquist/esphome-b2500 #4: Missing data points for the timers

ugath, 2024-06-29:
No data points for the timers (e.g. "b2500/%i/timer/1/enabled", "b2500/%i/timer/1/value"...) are created in iobroker.

**Additional context**

[image]

**Device information**

- ESP Board: ESP32

- B2500 Brand: Marstek

- B2500 Version: v2

**Configuration**

```json
{
  "name": "b2500",
  "friendly_name": "B2500",
  "poll_interval_seconds": 5,
  "mqtt": {
    "topic": "b2500",
    "broker": "192.168.178.150",
    "port": 1883,
    "username": "iobroker",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "FRITZ!Box 7490 UG",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "auto",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": true,
  "enable_cmd30": true,
  "enable_esp_temperature": false,
  "enable_powermeter": false,
  "enable_experimental_commands": true,
  "enable_hexdump": true,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": true,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "storages": [
    {
      "name": "B2500",
      "version": 1,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2024-06-29:
This is because you set the storage version to 1 instead of 2.

ugath, 2024-06-29:
> This is because you set the storage version to 1 instead of 2.

Sorry, my mistake. ;)

closed 2024-06-29 by tomquist