tomquist/esphome-b2500 #6: Timer Werte fehlen

RALFISBORN, 2024-06-30:
Hallo,

mir ist aufgefallen, dass im Gegensatz zu der Lösung von Nooke die Werte für die Zellspannung pro Zelle sowie die Value für den Zeitraum + Leistung fehlen.

Hier mal als Beispiel, wie es bei der anderen Lösung aussieht:
[image]

[image]

Kann das noch integriert werden?

tomquist, 2024-07-01:
Die Werte müssten jetzt auftauchen.

RALFISBORN, 2024-07-02:
Danke für das Update:
Die Timerwerte tauchen jetzt auf. 

Die Werte für die Zellspannung fehlen sowohl im Web als auch als MQTT (da tauchen sie im Broker gar nicht auf):
[image]

tomquist, 2024-07-02:
Hast du unter „Advanced“ den Schalter Enable Cellquery an?

RALFISBORN, 2024-07-02:
Ok Sorry mein Fehler. Mit der Aktivierung der Advanced Funktion klappts nun. Dankeschön :)

RALFISBORN, 2024-07-02:
Noch als Ergänzung für IOBroker Nutzer:
Die Timer (enable/disable) lassen sich nur über MQTT parametrieren, wenn im Adapter der Hacken bei:
[image]

nicht gesetzt ist.