tomquist/esphome-b2500 #8: keine wlan verbindung

denjo1982, 2024-07-03:
ich habe diesen ESP32

 ESP32 Entwicklungsplatine NodeMCU Module: ESP32 Typ C NodeMCU Development Board, ESP-WROOM-32, 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Integrated, CH340C Chip (2er)

und bekomme keine wlan verbinung aufgebaut sobald ich die bin datei installiere.

**Device information**

- ESP Board:  ESP32 Entwicklungsplatine NodeMCU Module: ESP32 Typ C NodeMCU Development Board, ESP-WROOM-32, 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Integrated, CH340C Chip (2er)

- B2500 Brand: Bluepalm

- B2500 Version: V1.2 (V2)

**Configuration**

```json
{
  "name": "b2500",
  "friendly_name": "b2500",
  "poll_interval_seconds": 5,
  "mqtt": {
    "topic": "b2500v2",
    "broker": "192.168.178.44",
    "port": "1324",
    "username": "denjo",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "Fritz!Box Denjo",
    "password": "***"
  },
  "board": "az-delivery-devkit-v4",
  "variant": "auto",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": true,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "storages": [
    {
      "name": "B2500",
      "version": 2,
      "mac_address": "***"
    },
    {
      "name": "BC2500",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2024-07-03:
Der ESP32 kann nur 2,4Ghz. Ist in dem WLAN 2,4Ghz aktiviert?

Was sagen die Logs des ESP32? Die kannst du wie folgt abrufen:
1. Gehe zu https://web.esphome.io/
2. Klick auf "Connect"
3. Gerät auswählen
4. Logs
5. Reset Device

Dann alles rauskopieren und hier einfügen.

denjo1982, 2024-07-03:
ja es ist 2,4Ghz wlan

`[12:32:36][W][wifi_esp32:663]: Event: Disconnected ssid='Fritz!Box Denjo' reason='Probe Request Unsuccessful'
[W][wifi:652]: Error while connecting to network.
[I][wifi:312]: WiFi Connecting to 'Fritz!Box Denjo'...
[W][wifi_esp32:663]: Event: Disconnected ssid='Fritz!Box Denjo' reason='Probe Request Unsuccessful'
[W][wifi:652]: Error while connecting to network.
ets Jun  8 2016 00:22:57
[12:32:36]
[12:32:36]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:32:36]configsip: 0, SPIWP:0xee
[12:32:36]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:32:36]mode:DIO, clock div:2
[12:32:36]load:0x3fff0030,len:6644
[12:32:36]load:0x40078000,len:15056
[12:32:36]ho 0 tail 12 room 4
[12:32:36]load:0x40080400,len:3628
[12:32:36]entry 0x40080668
[12:32:36]I (28) boot: ESP-IDF 4.4.7 2nd stage bootloader
[12:32:36]I (29) boot: compile time 19:55:08
[12:32:36]W (29) boot: Unicore bootloader
[12:32:36]I (31) boot: chip revision: v1.0
[12:32:36]I (35) boot.esp32: SPI Speed      : 40MHz
[12:32:36]I (40) boot.esp32: SPI Mode       : DIO
[12:32:36]I (44) boot.esp32: SPI Flash Size : 4MB
[12:32:36]I (49) boot: Enabling RNG early entropy source...
[12:32:36]I (54) boot: Partition Table:
[12:32:36]I (58) boot: ## Label            Usage          Type ST Offset   Length
[12:32:36]I (65) boot:  0 otadata          OTA data         01 00 00009000 00002000
[12:32:36]I (73) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[12:32:36]I (80) boot:  2 app0             OTA app          00 10 00010000 001c0000
[12:32:36]I (88) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[12:32:36]I (95) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[12:32:36]I (103) boot: End of partition table
[12:32:36]I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4f098h (323736) map
[12:32:36]I (233) esp_image: segment 1: paddr=0005f0c0 vaddr=3ffbdb60 size=00f58h (  3928) load
[12:32:36]I (234) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=114664h (1132132) map
[12:32:37]I (648) esp_image: segment 3: paddr=0017468c vaddr=3ffbeab8 size=038dch ( 14556) load
[12:32:37]I (654) esp_image: segment 4: paddr=00177f70 vaddr=40080000 size=1ce40h (118336) load
[12:32:37]I (717) boot: Loaded app from partition at offset 0x10000
[12:32:37]I (717) boot: Disabling RNG early entropy source...
[12:32:37]I (729) cpu_start: Unicore app
[12:32:37]I (729) cpu_start: Pro cpu up.
[12:32:37]I (729) cpu_start: Single core mode
[12:32:37]I (743) cpu_start: Pro cpu start user code
[12:32:37]I (743) cpu_start: cpu freq: 160000000
[12:32:37]I (743) cpu_start: Application information:
[12:32:37]I (747) cpu_start: Project name:     b2500
[12:32:37]I (752) cpu_start: App version:      2024.6.4
[12:32:37]I (757) cpu_start: Compile time:     Jul  2 2024 19:54:01
[12:32:37]I (763) cpu_start: ELF file SHA256:  5e55e51c6ccffebf...
[12:32:37]I (769) cpu_start: ESP-IDF:          4.4.7
[12:32:37]I (774) cpu_start: Min chip rev:     v0.0
[12:32:37]I (779) cpu_start: Max chip rev:     v3.99 
[12:32:37]I (784) cpu_start: Chip rev:         v1.0
[12:32:37]I (789) heap_init: Initializing. RAM available for dynamic allocation:
[12:32:37]I (796) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[12:32:37]I (802) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[12:32:37]I (808) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[12:32:37]I (814) heap_init: At 3FFCB8B8 len 00014748 (81 KiB): DRAM
[12:32:37]I (820) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[12:32:37]I (827) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
[12:32:37]I (833) heap_init: At 4009CE40 len 000031C0 (12 KiB): IRAM
[12:32:37]I (839) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
[12:32:37]I (846) spi_flash: detected chip: generic
[12:32:37]I (850) spi_flash: flash io: dio
[12:32:37]I (855) cpu_start: Starting scheduler on PRO CPU.`

tomquist, 2024-07-03:
Ich glaube du warst ein bisschen zu schnell. Kannst du das ganze nochmal probieren, aber nach dem Reset ein wenig warten bevor du die Logs raus kopierst?

denjo1982, 2024-07-03:
da kann ich leider lange warten da kommt nicht mehr

tomquist, 2024-07-03:
Ich kann leider auch nur raten. Würde sagen entweder das falsche Board genutzt, oder irgendein anderes Problem mit deinem WLAN. Habe die Diskussion gefunden: https://community.home-assistant.io/t/esphome-esp32-unable-to-connect-wifi/621494/18

Eventuell läuft dein WLAN auf Kanal 12,13 oder 14, womit es häufiger Probleme mit den ESP32 gibt.

denjo1982, 2024-07-03:
hatte verschiedene boards ausprobiert und jetzt mal dieses genommen:
upesy_wroom

log:

`[14:06:32]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:32]configsip: 0, SPIWP:0xee
[14:06:32]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:32]mode:QIO, clock div:1
[14:06:32]load:0x3fff0030,len:6644
[14:06:32]load:0x16ff6b25,len:-2071867686
[14:06:32]ets Jun  8 2016 00:22:57
[14:06:32]
[14:06:32]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:32]configsip: 0, SPIWP:0xee
[14:06:32]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:32]mode:QIO, clock div:1
[14:06:32]load:0x3fff0030,len:6644
[14:06:32]load:0x16ff6b25,len:-2071867686
[14:06:33]ets Jun  8 2016 00:22:57
[14:06:33]
[14:06:33]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:33]configsip: 0, SPIWP:0xee
[14:06:33]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:33]mode:QIO, clock div:1
[14:06:33]load:0x3fff0030,len:6644
[14:06:33]load:0x16ff6b25,len:-2071867686
[14:06:33]ets Jun  8 2016 00:22:57
[14:06:33]
[14:06:33]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:33]configsip: 0, SPIWP:0xee
[14:06:33]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:33]mode:QIO, clock div:1
[14:06:33]load:0x3fff0030,len:6644
[14:06:33]load:0x16ff6b25,len:-2071867686
[14:06:33]ets Jun  8 2016 00:22:57
[14:06:33]
[14:06:33]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:33]configsip: 0, SPIWP:0xee
[14:06:33]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:33]mode:QIO, clock div:1
[14:06:33]load:0x3fff0030,len:6644
[14:06:33]load:0x16ff6b25,len:-2071867686
[14:06:34]ets Jun  8 2016 00:22:57
[14:06:34]
[14:06:34]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:34]configsip: 0, SPIWP:0xee
[14:06:34]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:34]mode:QIO, clock div:1
[14:06:34]load:0x3fff0030,len:6644
[14:06:34]load:0x16ff6b25,len:-2071867686
[14:06:34]ets Jun  8 2016 00:22:57
[14:06:34]
[14:06:34]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:34]configsip: 0, SPIWP:0xee
[14:06:34]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:34]mode:QIO, clock div:1
[14:06:34]load:0x3fff0030,len:6644
[14:06:34]load:0x16ff6b25,len:-2071867686
[14:06:34]ets Jun  8 2016 00:22:57
[14:06:34]
[14:06:34]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:34]configsip: 0, SPIWP:0xee
[14:06:34]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:34]mode:QIO, clock div:1
[14:06:34]load:0x3fff0030,len:6644
[14:06:34]load:0x16ff6b25,len:-2071867686
[14:06:35]ets Jun  8 2016 00:22:57
[14:06:35]
[14:06:35]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:35]configsip: 0, SPIWP:0xee
[14:06:35]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:35]mode:QIO, clock div:1
[14:06:35]load:0x3fff0030,len:6644
[14:06:35]load:0x16ff6b25,len:-2071867686
[14:06:35]ets Jun  8 2016 00:22:57
[14:06:35]
[14:06:35]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:35]configsip: 0, SPIWP:0xee
[14:06:35]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:35]mode:QIO, clock div:1
[14:06:35]load:0x3fff0030,len:6644
[14:06:35]load:0x16ff6b25,len:-2071867686
[14:06:36]ets Jun  8 2016 00:22:57
[14:06:36]
[14:06:36]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:36]configsip: 0, SPIWP:0xee
[14:06:36]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:36]mode:QIO, clock div:1
[14:06:36]load:0x3fff0030,len:6644
[14:06:36]load:0x16ff6b25,len:-2071867686
[14:06:36]ets Jun  8 2016 00:22:57
[14:06:36]
[14:06:36]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:36]configsip: 0, SPIWP:0xee
[14:06:36]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:36]mode:QIO, clock div:1
[14:06:36]load:0x3fff0030,len:6644
[14:06:36]load:0x16ff6b25,len:-2071867686
[14:06:36]ets Jun  8 2016 00:22:57
[14:06:36]
[14:06:36]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:06:36]configsip: 0, SPIWP:0xee
[14:06:36]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:06:36]mode:QIO, clock div:1
[14:06:36]load:0x3fff0030,len:6644
[14:06:36]load:0x16ff6b25,len:-2071867686`

tomquist, 2024-07-03:
Mit dem scheint er nicht zu booten.

denjo1982, 2024-07-03:
bin leider kein profi:

was kann ich nu machen? ich hatte 3 bords bis jetzt probiert weiß auch nicht welches ich sonst nehmen sollte.

tomquist, 2024-07-03:
Hast du meine Nachricht oben schon gelesen bzgl des WLAN Kanals?

denjo1982, 2024-07-03:
läuft auf Kanal 10

tomquist, 2024-07-03:
Das einzige das mir noch einfällt ist, die config `output_power` mal zu setzen, wie in dem Link beschrieben. Dazu müsstest du aber die Config selbst anpassen und dann selbst bauen, da das mit dem Tool momentan nicht geht.

Wenn das nichts hilft, würde ich mich eventuell mal an die ESPHome community wenden.

denjo1982, 2024-07-03:
jetzt hab ich das wlan auf kanal 5 und gleicher fehler.

Das mir leider zu hoch, sonst muss ich gucken wo ich woanders einen esp her bekomme. weil bei amazon bekommt man nur die.
kannst du mir einen empfehlen?

robbe1912, 2024-07-05:
Seit heute hat meiner verbindungsabbrüche mit dem WLAN. Ich habe meine Batterie heute auf V255 geupdated. Kann das daran liegen?

knickohr, 2024-07-06:
Ich kann das Problem bestätigen. Es verliert irgendwie die TCP-Verbindung :

Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[13:44:21]I (209) esp_image: segment 1: paddr=0004bef0 vaddr=3ffbdb60 size=04128h ( 16680) load
[13:44:21]I (216) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=10541ch (1070108) map
[13:44:22]I (603) esp_image: segment 3: paddr=00155444 vaddr=3ffc1c88 size=006e4h (  1764) load
[13:44:22]I (604) esp_image: segment 4: paddr=00155b30 vaddr=40080000 size=1cde8h (118248) load
[13:44:22]I (671) boot: Loaded app from partition at offset 0x10000
[13:44:22]I (734) boot: Set actual ota_seq=1 in otadata[0]
[13:44:22]I (735) boot: Disabling RNG early entropy source...
[13:44:22]I (745) cpu_start: Unicore app
[13:44:22]I (745) cpu_start: Pro cpu up.
[13:44:22]I (746) cpu_start: Single core mode
[13:44:22]I (760) cpu_start: Pro cpu start user code
[13:44:22]I (760) cpu_start: cpu freq: 160000000
[13:44:22]I (760) cpu_start: Application information:
[13:44:22]I (764) cpu_start: Project name:     bc2500-1-ble-idf
[13:44:22]I (770) cpu_start: App version:      2024.6.6
[13:44:22]I (775) cpu_start: Compile time:     Jul  6 2024 13:10:52
[13:44:22]I (781) cpu_start: ELF file SHA256:  c36fd02a3c037d2f...
[13:44:22]I (787) cpu_start: ESP-IDF:          4.4.7
[13:44:22]I (792) cpu_start: Min chip rev:     v0.0
[13:44:22]I (797) cpu_start: Max chip rev:     v3.99 
[13:44:22]I (801) cpu_start: Chip rev:         v1.0
[13:44:22]I (806) heap_init: Initializing. RAM available for dynamic allocation:
[13:44:22]I (813) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[13:44:22]I (819) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[13:44:22]I (825) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[13:44:22]I (832) heap_init: At 3FFCB548 len 00014AB8 (82 KiB): DRAM
[13:44:22]I (838) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[13:44:22]I (844) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
[13:44:22]I (850) heap_init: At 4009CDE8 len 00003218 (12 KiB): IRAM
[13:44:22]I (857) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
[13:44:22]I (864) spi_flash: detected chip: generic
[13:44:22]I (868) spi_flash: flash io: dio
[13:44:22]I (873) cpu_start: Starting scheduler on PRO CPU.
[13:44:22][I][logger:156]: Log initialized
[13:44:22][I][app:029]: Running through setup()...
[13:44:22][I][main:2054]: Setting Discharge Threshold on device 1 to 0
[13:44:22][I][main:4123]: Sending command to device 1: [12] 0
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][main:2034]: Setting DOD on device 1 to 0
[13:44:22][I][main:2054]: Setting Discharge Threshold on device 2 to 0
[13:44:22][I][main:4123]: Sending command to device 2: [12] 0
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][main:2034]: Setting DOD on device 2 to 0
[13:44:22][I][main:4123]: Sending command to device 1: [14] 0
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][main:4123]: Sending command to device 1: [14] 0
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][main:4123]: Sending command to device 2: [14] 0
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][main:4123]: Sending command to device 2: [14] 0
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][main:2019]: Setting PV2 Passthrough on device 1 to 1
[13:44:22][I][main:4123]: Sending command to device 1: [13] 1
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][main:2019]: Setting PV2 Passthrough on device 2 to 1
[13:44:22][I][main:4123]: Sending command to device 2: [13] 1
[13:44:22][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:22][I][esp-idf:000]: I (1265) wifi:
[13:44:22][I][esp-idf:000]: wifi driver task: 3ffdf728, prio:23, stack:6656, core=0
[13:44:22][I][esp-idf:000]: 
[13:44:22]
[13:44:22][I][esp-idf:000][wifi]: I (1271) system_api: Base MAC address is not set
[13:44:22]
[13:44:22][I][esp-idf:000][wifi]: I (1280) system_api: read default base MAC address from EFUSE
[13:44:22]
[... 125 lines cut ...]
[13:44:25][I][esp-idf:000][wifi]: 
[13:44:25]
[13:44:26][I][esp-idf:000][sys_evt]: I (4834) esp_netif_handlers: sta ip: 192.168.100.116, mask: 255.255.255.0, gw: 192.168.100.1
[13:44:26]
[13:44:26][I][wifi:616]: WiFi Connected!
[13:44:26][W][component:157]: Component mqtt set Warning flag: unspecified
[13:44:26][I][mqtt:234]: Connecting to MQTT...
[13:44:26][I][esp-idf:000][wifi]: I (4866) wifi:
[13:44:26][I][esp-idf:000][wifi]: <ba-add>idx:1 (ifx:0, dc:39:6f:28:1a:27), tid:0, ssn:0, winSize:64
[13:44:26][I][esp-idf:000][wifi]: 
[13:44:26]
[13:44:26][W][component:170]: Component wifi cleared Warning flag
[13:44:26][W][component:170]: Component mqtt cleared Warning flag
[13:44:26][I][mqtt:274]: MQTT Connected!
[13:44:27][I][app:062]: setup() finished successfully!
[13:44:27][I][app:100]: ESPHome version 2024.6.6 compiled on Jul  6 2024, 13:39:21
[13:44:27][I][main:4123]: Sending command to device 1: [14] 2
[13:44:27][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:44:27][I][main:4123]: Sending command to device 2: [14] 2
[13:44:27][W][ble_client.automation:141]: Cannot write to BLE characteristic - not connected
[13:45:02][W][ble_sensor:123]: [B2500 - 1 - 1: Bluetooth Low Energy Client] Cannot poll, not connected
[13:45:18][W][ble_sensor:123]: [B2500 - 2 - 2: Bluetooth Low Energy Client] Cannot poll, not connected
[13:45:22][I][safe_mode:041]: Boot seems successful; resetting boot loop counter

Danach ist Schluß.

tomquist, 2024-07-06:
Hm, das Problem von @knickohr sieht für mich anders aus. Eventuell würde es Sinn machen, das in einem separaten Issue zu diskutieren. Mich würde deine config interessieren. Sieht erstmal so aus als hättest du zwei v1 Speicher, ist das korrekt? Was mir auch noch auffällt, dass der ESP32 sofort nach dem Boot versucht etwas an den Bluetooth client zu senden. Kann es sein, dass in deinem MQTT broker folgende topics auf "Retain" gesetzt sind?
- `b2500/1/battery/dod/set`
- `b2500/2/battery/dod/set`
- `b2500/1/battery/discharge_threshold/set`
- `b2500/2/battery/discharge_threshold/set`
- `b2500/1/pv2/passtrough/set`
- `b2500/2/pv2/passtrough/set`
Idealerweise sollte auf die `/set` topics nicht mit dem "retain" flag geschrieben werden, sonst versucht der ESP32 bei jedem reconnect die Werte neu zu schreiben.

knickohr, 2024-07-06:
Korrekt, da hängen 2 BP2500 dran.

und auch richtig, sämtliche Topics sind retained geflaggt, das kommt von dem „alten“ yaml von noone2k. Hab ich ihm auch schon gesagt das dies irgendwie Dumm ist ☹️

tomquist, 2024-07-06:
Ich rede aber speziell über die `/set` topics. Auf die schreibt weder das yaml von noone2k, noch das yaml des config tools. Die Topics werden nur zum Steuern genutzt und ESPHome subscribed sich darauf nur.

tomquist, 2024-07-06:
Du kannst mal versuchen die `/set` Topics mal über den MQTT Explorer zu löschen. Eventuell hilft das ja.

knickohr, 2024-07-06:
Ich habe gerade die mosquitto Datenbank der retained Topics gelöscht. Mal sehen wann sie wieder da sind 😉

knickohr, 2024-07-06:
Bist Du sicher das die Set vom MQTT kommen ?  Wifi ist ja noch gar nicht verbunden 😳

tomquist, 2024-07-07:
Hm, guter Punkt! Sieht so aus als käme das dann von woanders her. Vermutlich von der `restore_value` setting.

Ich habe eben mal noch mehr logs gepushed. Darüber sollte man sehen, ob das reguläre polling läuft. Ansonsten sehen deine Logs (@knickohr) erstmal relativ gut aus. Ist der ESP32 in Reichweite des Speichers? Ist die eingegebene MAC Adresse korrekt? Meine Logs sehen exakt so aus wie deine, wenn sich der ESP nicht in der Nähe des Speichers befindet.

knickohr, 2024-07-07:
Ja, aber es sollte sich doch mit dem WLAN weiterhin verbinden und nicht einfach stehen bleiben. Hab ein Ping mitlaufen lassen. Tut erst, bis es dann an der Stelle stehen bleibt, dann hört auch der Ping auf 😲

so als würde es an der Stelle auf was warten und einfach nicht weiter machen.

knickohr, 2024-07-07:
Hier mal ein MQTT-Log während des Starten. Es verbindet sich also definitiv kurz mit dem WLAN und auch dem Broker. Aber aus (mir) unerfindlichen Gründen trennt es sich dann mit TCP disconnect.

[image]

tomquist, 2024-07-07:
Sind die Logs von oben direkt über die serielle Schnittstelle ausgelesen? Ich tippe auf einen Crash des ESP32. Den sollte man aber normalerweise in den Logs der seriellen Schnittstelle sehen.

knickohr, 2024-07-07:
Bei, das ist das MQTT-Log

knickohr, 2024-07-07:
Nein, das ist das MQTT-Log.

Das serielle ist von gestern.