tomquist/esphome-b2500 #9: Mehrere kleine Issues

knickohr, 2024-07-06:
Building über den Button funktioniert soweit. Alllerdings fielen mir 2 Kleinigkeiten auf, so das ich manuel "gebuildet" habe.

Temperatursensor :
Kann natürlich nur ausgewählt werden wenn die Hardware das auch unterstützt. Hier sollte zuvor eine Abfrage eingebaut sein, um die Checkbox zu enablen/disablen.

sensor.adc: [source config1.yaml:814]
  platform: adc
  id: temp_analog_reading
  
  ESP32 doesn't support ADC on this pin.
  pin: GPIO3
  attenuation: 11db

Ein weitere Punkt der offenbar gravierender ist :
Das Tool generiert eine Zeile in der yaml mit

datetime:

Ohne weitere Konfiguration. Beim manuellen Erstellen des Bins schägt es hier fehl.

failed config

datetime: [source config1.yaml:349]
  
  Component not found: datetime.

Nachdem ich diese eine Zeile gelöscht habe lief es fehlerfrei durch

tomquist, 2024-07-06:
Danke fürs Melden. Das Problem mit dem leeren `datetime` ist behoben. Das Problem mit `ota` kommt daher, dass ESPHome ein breaking change im letzten Release eingebaut hat. In der neusten ESPHome Version ist `ota` ein Array, kein Dictionary mehr. Das Problem lässt sich durch ein Update von ESPHome beheben.

Das Problem mit dem Temperatur-Sensor ist nicht einfach zu lösen, da die App keine Informationen über die Boards und deren Hardware hat. Würde das auch ungern einbauen. Eventuell werde ich eine Warnung anzeigen, wenn der Haken gesetzt ist.

tomquist, 2024-07-06:
Habe noch ein kleines Update gemacht, das standardmäßig den internen Temperatur-Sensor des ESP32 chips verwendet, anstatt den NTC Thermistor zu verwenden.

closed 2024-07-08 by tomquist