tomquist/esphome-b2500 #11: WLAN-Verbindung bricht ab, kein Web-GUI

knickohr, 2024-07-08:
OK,
da es hier bei mir um ein seltsames Problem geht das sich der ESP nur kurz mit dem WiFi verbindet, wie auch MQTT, mache ich einen neuen Issue auf. Es verbindet sich kurz nach dem Start, danach verstummt es bis zum nächsten (automatischen) Reboot Das Webinbterface kann nie aufgerufen werden, über MQTT und Serial sehe ich aber das Debig.

**Device information**

- ESP Board: [e.g. ESP32-C3]
ESP32 DEV-Kit-C AZ-Delivery

- B2500 Brand: [e.g. Marstek]
BluePalm (Hornbach)

- B2500 Version: [e.g. v1, v2, v3]
V1

**Configuration**

```json
{
  "name": "bc2500-1-ble-idf",
  "friendly_name": "B2500-1",
  "poll_interval_seconds": "15",
  "mqtt": {
    "topic": "b2500-1",
    "broker": "192.168.100.80",
    "port": 1883,
    "username": "mqttuser",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "Odyssee2001",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "auto",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": true,
  "enable_timer_query": false,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": true,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": false,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "storages": [
    {
      "name": "1",
      "version": 1,
      "mac_address": "***"
    },
    {
      "name": "2",
      "version": 1,
      "mac_address": "***"
    }
  ],
  "flash_size": "4MB",
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_dio_flash_mode": false
}
```
Hier mal ein etwas längeres Log :
[Log.txt](https://github.com/user-attachments/files/16124683/Log.txt)

Interessant finde ich das MQTT immer wieder .../bluetooth/enabled OFF bringt.

Die Speicher sind während dem MQTT-Log in Reichweite, jedoch nicht beim seriellen Log am PC.

Vielleicht kann man was sehen.

tomquist, 2024-07-08:
Kannst du mal versuchen den Webserver auszuschalten? Die ESP32-C3 sind ziemlich schwach auf der Brust. Ich habe das ganze auf einem Shelly PM Mini laufen, in dem der gleiche ESP32 verbaut ist und musste den Webserver abschalten, damit der nicht abstürzt.

knickohr, 2024-07-08:
Finde ich jetzt spannend !  Mit der yaml von noone2k funzt es ja.

Aber OK, ich teste.

knickohr, 2024-07-08:
Tatsächlich !!!

[Log.txt](https://github.com/user-attachments/files/16125784/Log.txt)

Ich teste es jetzt mal mit Verbindung zum Speicher.

knickohr, 2024-07-08:
Alles fein. Frage mich halt was bei deiner Version anders ist als bei noone2k, außer das es „maschinell“ erstellt wird 🤔

knickohr, 2024-07-08:
Im Log im ersten Beitrag ist einer drin :

[09:49:38]abort() was called at PC 0x401b6daa on core 0
[09:49:38]
[09:49:38]
[09:49:38]Backtrace: 0x4008247e:0x3ffed460 0x40091495:0x3ffed480 0x4009707a:0x3ffed4a0 0x401b6daa:0x3ffed510 0x401b595b:0x3ffed530 0x401b58c2:0x3ffed550 0x400d529d:0x3ffed570 0x400d535a:0x3ffed590 0x400d5399:0x3ffed5c0 0x400fa1a1:0x3ffed5e0 0x400fab59:0x3ffed600 0x40123e7a:0x3ffed620 0x40123e17:0x3ffed640
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008247e: panic_abort at /home/ubuntu/.platformio/packages/framework-espidf/components/esp_system/panic.c:408
WARNING Decoded 0x40091495: esp_system_abort at /home/ubuntu/.platformio/packages/framework-espidf/components/esp_system/esp_system.c:137
WARNING Decoded 0x4009707a: abort at /home/ubuntu/.platformio/packages/framework-espidf/components/newlib/abort.c:46
WARNING Decoded 0x401b6daa: __wrap__Unwind_Resume_or_Rethrow
WARNING Decoded 0x401b595b: __cxa_throw at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:90
WARNING Decoded 0x401b58c2: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x400d529d: __gnu_cxx::new_allocator<esphome::esp32_ble::BLEEvent*>::allocate(unsigned int, void const*) at /home/ubuntu/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/ext/new_allocator.h:111
 (inlined by) std::allocator_traits<std::allocator<esphome::esp32_ble::BLEEvent*> >::allocate(std::allocator<esphome::esp32_ble::BLEEvent*>&, unsigned int) at /home/ubuntu/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/alloc_traits.h:436
 (inlined by) std::_Deque_base<esphome::esp32_ble::BLEEvent*, std::allocator<esphome::esp32_ble::BLEEvent*> >::_M_allocate_node() at /home/ubuntu/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_deque.h:603
 (inlined by) void std::deque<esphome::esp32_ble::BLEEvent*, std::allocator<esphome::esp32_ble::BLEEvent*> >::_M_push_back_aux<esphome::esp32_ble::BLEEvent* const&>(esphome::esp32_ble::BLEEvent* const&) at /home/ubuntu/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/deque.tcc:488
WARNING Decoded 0x400d535a: std::deque<esphome::esp32_ble::BLEEvent*, std::allocator<esphome::esp32_ble::BLEEvent*> >::push_back(esphome::esp32_ble::BLEEvent* const&) at /home/ubuntu/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_deque.h:1562
 (inlined by) std::queue<esphome::esp32_ble::BLEEvent*, std::deque<esphome::esp32_ble::BLEEvent*, std::allocator<esphome::esp32_ble::BLEEvent*> > >::push(esphome::esp32_ble::BLEEvent* const&) at /home/ubuntu/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_queue.h:252
 (inlined by) esphome::esp32_ble::Queue<esphome::esp32_ble::BLEEvent>::push(esphome::esp32_ble::BLEEvent*) at /home/ubuntu/hm2500pub/config/.esphome/build/bc2500-1-ble-idf/src/esphome/components/esp32_ble/queue.h:30
WARNING Decoded 0x400d5399: esphome::esp32_ble::ESP32BLE::gap_event_handler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at /home/ubuntu/hm2500pub/config/.esphome/build/bc2500-1-ble-idf/src/esphome/components/esp32_ble/ble.cpp:319
WARNING Decoded 0x400fa1a1: btc_gap_ble_cb_to_app at /home/ubuntu/.platformio/packages/framework-espidf/components/bt/host/bluedroid/btc/profile/std/gap/btc_gap_ble.c:70
WARNING Decoded 0x400fab59: btc_gap_ble_adv_pkt_handler at /home/ubuntu/.platformio/packages/framework-espidf/components/bt/host/bluedroid/btc/profile/std/gap/btc_gap_ble.c:625
WARNING Decoded 0x40123e7a: osi_thread_generic_event_handler at /home/ubuntu/.platformio/packages/framework-espidf/components/bt/common/osi/thread.c:425
 (inlined by) osi_thread_generic_event_handler at /home/ubuntu/.platformio/packages/framework-espidf/components/bt/common/osi/thread.c:418
WARNING Decoded 0x40123e17: osi_thread_run at /home/ubuntu/.platformio/packages/framework-espidf/components/bt/common/osi/thread.c:165

BuFisch, 2024-07-11:
Hi,
ich habe das gleiche Problem mit dem Wlan Abbrüche, habe zwei unterschiedlichen ESP32 versucht (ESP-32 Dev Kit C V4).
Meine Frage ist: welches ESP32 Board ist das beste für dieses Projekt?
Ich finde dieses Projekt genial weil man sehr einfach eine Konfiguration erstellen kann, kleiner Wehrmutstropfen gibt es, man nach der Erstellung in ESPHome die Gänsefüßchen bei WiFi Passwort und SSID entfernen wenn man mit secrets.yaml arbeitet.
Ich habe mich auch schon mit der yaml von noone2k versucht aber scheitere auch mit WiFi Abbrüche bzw. der ESP32 hängt sich auf.
Ich habe von BE COOL den BC-B2500 v1 und die Erweiterung BC-P2500.
Grüße Burkhard

tomquist, 2024-07-11:
Ich habe mit diesem hier gute Erfahrungen gemacht: https://www.amazon.de/dp/B0C6KN35L2?psc=1&ref=ppx_yo2ov_dt_b_product_details

Das board ist etwas Leistungsfähiger und hat daher keine Probleme mit crashes. Zu beachten ist, dass mit diesem board der Haken bei "Enable DIO Flash Mode" gesetzt werden muss.

BuFisch, 2024-07-11:
Danke, ist bestellt.
Bei Board bleibt esp32dev? und Variant bleibt auto?

tomquist, 2024-07-11:
Nein, da es ein ESP32-s3 ist, muss board auf `esp32-s3-devkitc-1` gesetzt werden. Du kannst mal `variant` auf `auto` setzen und wenn das nicht geht, dann sollte `variant` auf `esp32s3` gesetzt werden. Außerdem hat das Teil 16MB ram, du kannst also die flash size auf 16MB stellen (wobei das glaube ich nicht zwingend notwendig ist).

knickohr, 2024-07-11:
Ohne jetzt sarkastisch zu klingen und irgendjemand schlecht zu machen :  Muß das wirklich ein S3 N16R8 sein ?

Das ist ja schon ein gewaltiger Brummer !

Und ja, die generierte Version bootete in 2 Tagen öfters als die „alte“ noone2k-yaml über mehrere Wochen. Ich werde aber trotzdem mal ein S3 besorgen, möchte nur gerne einen mit externer Antenne in es nochmal versuchen. Momentan bin ich halt wieder zurück auf der alten.

tomquist, 2024-07-11:
Ja, wie gesagt enthält das generierte Script einige Dinge, die es in der config von noone2k nicht gibt, von dem her ist das nicht 1:1 vergleichbar. Man muss auch dazu sagen, dass die die config so wie sie ist auch schon extrem groß ist und gleich 2 Komponenten nutzt, bei denen ESPHome offiziell warnt, dass sie sehr viel Speicher benötigen (BLE stack & webserver) von dem her ist es nicht verwunderlich, dass die kleineren Boards das nicht verkraften. Ist auch ein Kompromiss zwischen mehr features und 1,50€ mehr bezahlen für ein größeres Board.

Ich kann mal versuchen die cell voltages aus den entities wieder zu entfernen und stattdessen direkt in MQTT zu schreiben, evtl. zusammengefasst in ein JSON object. Würde mich auch mal interessieren, ob es hilft, die cell queries auszuschalten. Eventuell kann das mal jemand testen.

knickohr, 2024-07-11:
Nicht falsch verstehen, es geht hier nicht um 1,50€ oder so.

Es hat mich halt erschreckt das die alte yaml mit Webserver und Cellquery stabiler rennt als die generierte Version wo ich bereits den Webserver abgeschaltet habe. Aber wie Du schon schreibst, es wird wohl an den Entities liegen.

Ich würde auch die anderen Speicherdaten in ein weiteres JSON schreiben 😉

azazul1980, 2024-07-12:
Ja bitte abändern mein Bord steigt doch öfters aus am tag

tomquist, 2024-07-12:
Ich kann mir das anschauen, sobald ich eine Bestätigung habe, dass das Abschalten der cell-queries die Abstürze verhindert. Will erstmal nicht so viel Zeit investieren, wenn es am Ende das Problem nicht beseitigt.

azazul1980, 2024-07-12:
Sorry die Cellquery ist bei mir gar nicht aktiv. Ich habe nur die Timer Aktiviert und das OTA.

tomquist, 2024-07-13:
@azazul1980 Ist bei dir der Webserver aktiv?

azazul1980, 2024-07-13:
Ne habe ich nicht 
[image]

azazul1980, 2024-07-13:
[image]

azazul1980, 2024-07-13:
[image]

BuFisch, 2024-07-13:
also mit dem empfohlenen Board läuft bei mir jetzt alles bestens, habe nach etwa 9 Stunden einen reboot des Speichers über WEB angeschubst und hat sich sofort wieder connected kein einen Abbruch oder Hänger

BuFisch, 2024-07-13:
[image]

tomquist, 2024-08-03:
Habe die letzten Wochen einige Zeit investiert und eine native ESPHome Komponente implementiert. Diese unterstützt die gleichen Features wie die von noone2k, nutzt aber eine komplett eigene Implementierung des B2500 Bluetooth Protokolls und basiert nicht mehr auf dem Code von noone2k. Die neue Version ist deutlich speichersparender und das reboot Problem sollte damit größtenteils behoben sein

Im [Config Generator](https://tomquist.github.io/esphome-b2500/) kann jetzt ganz oben die neue Version gewählt werden.
[image]

Ich habe versucht die MQTT Topics größtenteils gleich zu belassen wie in v1, es gibt aber ein paar wenige Unterschiede. Das betrifft insbesondere:
- Der Modus transparent/zuerst laden etc wird jetzt über den topic `charge_mode` herausgegeben, anstatt eines Switch.
- Die Zellspannung wird jetzt als JSON Objekt gesendet

Das ganze ist aktuell noch wenig getestet, daher würde ich mich über Rückmeldungen freuen!

azazul1980, 2024-08-03:
Cool,  ich würde evtl. testen bin aber die nächsten Wochen im Urlaub. Muss auch sagen das generell das ganze Stabil läuft. Die Reboots Kammern bei mir immer wenn der Timer mit dem Ausgabewert verändert wurde.

azazul1980, 2024-08-28:
@tomquist 

hallo,

Ich habe nun die neue Version getestet, bin aber wieder zurück gesprungen, da mir Werte gefehlt haben wie zum Beispiel der Akkus State den habe ich bei mir aktiv eingebaut. In einer Automatisierung wie ich die einzelnen Zerspanung auslesen kann, war mir auch nicht wirklich ersichtlich.

noone2k, 2024-08-28:
muss hier mal in die runde fragen, zu den problem-boards: sind das hauptsächlich esp32-devkit-c4 ? ( bspw. von az-delivery ? ).

hatte auch eines davon, welches dann aber nach ner weile ausstieg und immer wieder probleme  machte.
nur das abschalten einiger funktionenen liess der ding halbwegs stabil agieren.
für die meisten sachen sind die okay ... aber bei concurrent ble/wifi scheint es da zu hapern ...

seitdem ich auf ein anderes board umgestiegen bin, läuft das wesentlich besser ( in (fast) allen punkten ) ...
mittlweile läuft bei mir das meiste auf MH-ET Live boards ( sind ne art mini-esp32, hatte da noch "zufällig" nen 10er pack von ali rumliegen ) ...

azazul1980, 2024-08-28:
@noone2k 

Ja bei mir war es der Esp32 devkit c4.

Dein ESP32 Mini Board würde auch mit Tomquist seiner Lösung laufen?

knickohr, 2024-08-28:
Interessant !

Normalerweise sind die AZ-Dinger recht zuverlässig, zumindest ohne BLE. Ja, ich hatte zuerst auch ein Dev-Kit C, aber ein V3. Dann habe ich wegen der externen Antenne auf das hier gewechselt :

https://www.amazon.de/MELIFE-ESP32-DevKitC-Entwicklungsboard-ESP32-WROOM-32D-WROOM-32D/dp/B0811KLGDD

Da ist es aber genauso mau.

noone2k, 2024-08-28:
sollte mit jedem esp32 laufen, egal welche variante, sobald es vom framework unterstützt wird.

hatte das auch schon auf dem c6  ( esp32-c6-devkitc-1 ). ist aber genauso overkill, wie der h2 :)
man nimmt halt das was irgendwie in reichweite ist :grin: 

achja, da war ein typo meinerseits: devkit-*v4* ... manchmal kommt man da schon durcheinander ... :grin: 

gerade bei devkit nachbauten kann es immer zu produktions problemen kommen.
az-delivery stellt die ja auch nicht selbst her, sondern lässt nur ihr logo drauflasern ...
mal hat man glück und die produktion klappt und manchmal eben nicht,
ist bei china-clonen leider nicht auszuschließen ...

ganz gut fährt man immer mit den originalen, aber die sind entweder frühzeitig ausverkauft 
oder werden aufgrund neuerer produkte nicht mehr produziert.
da muss man glück haben, das sich jmd. 100% an das referenz-design von espressif hält oder
das die produktion einwandfrei funktioniert oder die fehler im referenz-design korrigiert wurden.

noone2k, 2024-08-28:
@knickohr 

mal bei ali nach mh-et live esp32 suchen .... ich bestelle öfters bei dem händler "feiwang" oder so ähnlich.
bisher zuverlässig und funktioniert BISHER immer.

ich such das mal raus ... 

ggf., falls jmd. noch nen esp32-mini von az-delivery rumliegen hat, damit mal probieren, dürfte das gleiche design sein ...
( hier trifft aber auch wieder der letze absatz des vorherigen posts zu ) ....

knickohr, 2024-08-28:
Mein Overkill-Board wäre noch ein Fusion-Board. Das ist von der Ahoy/OpenDTU Community selbst entwickeltes Referenzboard wo wir peinlichst auf genau solche Sachen geachtet haben.

Das möchte ich aber nicht als blankes ESP32-S3 verbraten, zumal da auch noch andere HW-Komponenten mit drauf sind die dann mit dem Script brach liegen.

azazul1980, 2024-08-28:
Ich habe mir den hier geholt und hatte heute einen Hänger damit aber sonnst 1 Monat ohne Probleme durchgelaufen.

https://amzn.eu/d/3K5TZ3v

knickohr, 2024-08-28:
Ja, das ist ein S3, auch Overkill. Aber naja, muß mal nac/ einem mit externer Antenne suchen gehen.

Flundi, 2024-09-02:
Hallo,
habe, glaub ich, das selbe Problem...
ESP32-WROOM-V2

[19:58:56]ets Jul 29 2019 12:21:46
[19:58:56]
[19:58:56]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:58:56]configsip: 0, SPIWP:0xee
[19:58:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:58:56]mode:DIO, clock div:2
[19:58:56]load:0x3fff0030,len:6644
[19:58:56]load:0x40078000,len:15056
[19:58:56]ho 0 tail 12 room 4
[19:58:56]load:0x40080400,len:3628
[19:58:56]entry 0x40080668
[19:58:56]I (29) boot: ESP-IDF 4.4.7 2nd stage bootloader
[19:58:56]I (29) boot: compile time 17:54:35
[19:58:56]W (29) boot: Unicore bootloader
[19:58:56]I (32) boot: chip revision: v3.0
[19:58:56]I (36) boot.esp32: SPI Speed      : 40MHz
[19:58:56]I (41) boot.esp32: SPI Mode       : DIO
[19:58:56]I (45) boot.esp32: SPI Flash Size : 4MB
[19:58:56]I (50) boot: Enabling RNG early entropy source...
[19:58:56]I (55) boot: Partition Table:
[19:58:56]I (59) boot: ## Label            Usage          Type ST Offset   Length
[19:58:56]I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
[19:58:56]I (73) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[19:58:56]I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
[19:58:56]I (88) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[19:58:57]I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[19:58:57]I (103) boot: End of partition table
[19:58:57]I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=59210h (365072) map
[19:58:57]I (248) esp_image: segment 1: paddr=00069238 vaddr=3ffbdb60 size=048c4h ( 18628) load
[19:58:57]I (256) esp_image: segment 2: paddr=0006db04 vaddr=40080000 size=02514h (  9492) load
[19:58:57]I (260) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=122880h (1190016) map
[19:58:57]I (693) esp_image: segment 4: paddr=001928a8 vaddr=40082514 size=1a850h (108624) load
[19:58:57]I (752) boot: Loaded app from partition at offset 0x10000
[19:58:57]I (752) boot: Disabling RNG early entropy source...
[19:58:57]I (764) cpu_start: Unicore app
[19:58:57]I (764) cpu_start: Pro cpu up.
[19:58:57]I (764) cpu_start: Single core mode
[19:58:57]I (778) cpu_start: Pro cpu start user code
[19:58:57]I (778) cpu_start: cpu freq: 160000000
[19:58:57]I (778) cpu_start: Application information:
[19:58:57]I (783) cpu_start: Project name:     b2500
[19:58:57]I (787) cpu_start: App version:      2024.8.1
[19:58:57]I (792) cpu_start: Compile time:     Sep  2 2024 17:53:36
[19:58:57]I (798) cpu_start: ELF file SHA256:  46efc27ba7cc2d74...
[19:58:57]I (804) cpu_start: ESP-IDF:          4.4.7
[19:58:57]I (809) cpu_start: Min chip rev:     v0.0
[19:58:57]I (814) cpu_start: Max chip rev:     v3.99 
[19:58:57]I (819) cpu_start: Chip rev:         v3.0
[19:58:57]I (824) heap_init: Initializing. RAM available for dynamic allocation:
[19:58:57]I (831) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[19:58:57]I (837) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[19:58:57]I (843) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[19:58:57]I (849) heap_init: At 3FFCC098 len 00013F68 (79 KiB): DRAM
[19:58:57]I (855) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[19:58:57]I (862) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
[19:58:57]I (868) heap_init: At 4009CD64 len 0000329C (12 KiB): IRAM
[19:58:57]I (874) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
[19:58:57]I (881) spi_flash: detected chip: generic
[19:58:57]I (885) spi_flash: flash io: dio
[19:58:57]I (891) cpu_start: Starting scheduler on PRO CPU.
[19:59:58][I][logger:156]: Log initialized
[I][app:029]: Running through setup()...
[I][esp-idf:000]: I (1134) wifi:
[I][esp-idf:000]: wifi driver task: 3ffd8ee4, prio:23, stack:6656, core=0
[I][esp-idf:000]: 

[... 241 lines cut ...]
[I][esp-idf:000][wifi]: I (61397) wifi:
[I][esp-idf:000][wifi]: Total power save buffer number: 16
[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: I (61408) wifi:
[I][esp-idf:000][wifi]: Init max length of beacon: 752/752
[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: I (61429) wifi:
[I][esp-idf:000][wifi]: Init max length of beacon: 752/752
[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: I (61442) wifi:
[I][esp-idf:000][wifi]: Total po
[19:59:58]assert failed: sntp_setoperatingmode IDF/components/lwip/lwip/src/apps/sntp/sntp.c:724 (Operating mode must not be set while SNTP client is running)
[19:59:58]
[19:59:58]
[19:59:58]Backtrace: 0x4008220e:0x3ffd5390 0x400913b9:0x3ffd53b0 0x400970ed:0x3ffd53d0 0x4013ccf4:0x3ffd54f0 0x4013a862:0x3ffd5510 0x4013c879:0x3ffd5530
[19:59:58]
[19:59:58]
[19:59:58]
[19:59:58]
[19:59:58]ELF file SHA256: 46efc27ba7cc2d74
[19:59:58]
[19:59:58]Rebooting...

closed 2026-03-15 by tomquist