tomquist/esphome-b2500 #12: Feature Request :  MQTT

knickohr, 2024-07-09:
Zwei weitere „Schmankerl“ für MQTT wären schön :
- Ausgabe der Daten optional als JSON-Payload für die Speicher und Zellspannungen. Würde den Topicwulst erheblich übersichtlicher machen
- Debug-Level Ausgabe über MQTT einstellbar, info, debug, error. Eine ständige Ausgabe der Infomeldungen halte ich über MQTT etwas zu übertrieben. Hier reicht wirklich error. Über Seriell natürlich komplett.

knickohr, 2024-07-10:
Erweiterung :

Ein Reboot wenn die MQTT-Verbindung nach x Minuten Disconnet nicht wieder zustande kommt wäre auch nicht schlecht. Ist bereits die zweite Nacht in der beide (habe 4 Batterien räumlich getrennt) ESP ausgestiegen ist.

Und nein, es liegt nicht am Broker, da meine Dutzende anderen Anwendungen weiterhin gelaufen sind 😉

tomquist, 2024-08-05:
> Ausgabe der Daten optional als JSON-Payload für die Speicher und Zellspannungen. Würde den Topicwulst erheblich übersichtlicher machen
Das habe ich mir auch schon überlegt, das ganze in ein großes JSON zu legen, oder zumindest die verschiedenen Speicher-Nachrichten (runtime info, cell queries, timer info, device info) in jeweils ein JSON topic zu packen. Werde evtl. verschiedene Templates anbieten, so dass man die Wahl hat.

> Debug-Level Ausgabe über MQTT einstellbar, info, debug, error. Eine ständige Ausgabe der Infomeldungen halte ich über MQTT etwas zu übertrieben. Hier reicht wirklich error. Über Seriell natürlich komplett.

Das wird nicht möglich sein. Das Debug-Level ist in das image "eingebrannt",  und kann nur zur Compile-Zeit verändert werden.

> Und noch eine Frage : Ist es möglich dem Ganzen ein Display zu spendieren. Z. B. um die Ausgangsleistung, etc. Anzeigen zu lassen ? Oder den Status des Speichers …
Habe da selbst kein Interesse daran, daher werde ich dafür keine Zeit investieren. Nehme aber gerne Pull Requests dafür an.

knickohr, 2024-08-05:
Nein, bitte nicht genau den umgekehrten Fall machen und alles in einen riesigen JSON-Blop knallen. Das wäre dann wieder übers Ziel hinaus geschossen. Ich wurde das Themenspezifisch zusammenfassen.

Beispielsweise je Speicher ein Topic mit
- Zellenspanungen (haben wir ja schon
- Die restlichen Batteriewerte (Kapazität, SOC/Prozent, Temperaturen)
- Die Stati des Speichers/Batterie/PV

Und eben noch den Gesundheitszustand des ESP (Temperatur, BLE, WiFi, Uptime)

Das mit dem Debug Level war auch so gemeint das man ihn beim Erstellen des yaml einmal in den Generator eingibt. Aber ich brauche halt nicht immer Info wenn das Ding mal sauber läuft. Da reicht mir Error.

OK, Display ist halt so eine Sache.  Eine kleine Anzeige wäre trotzdem fein. Z.B. für die Gesamtleiszung oder den Energiegehalt der Speicher. Wenn ich das richtig gesehen habe scheint das mit ESPhome recht einfach zu gehen. Viel Arbeit würde ich da auch nicht rein stecken.

tomquist, 2024-12-20:
> Das mit dem Debug Level war auch so gemeint das man ihn beim Erstellen des yaml einmal in den Generator eingibt. Aber ich brauche halt nicht immer Info wenn das Ding mal sauber läuft. Da reicht mir Error.

Das gibt es schon. Man kann aktuell zwischen INFO und DEBUG wechseln. Bei INFO sollte so gut wie kein log mehr erzeugt werden. DEBUG ist dagegen sehr verbose, und sollte wirklich nur fürs debugging angeschaltet werden.

HelmutFi, 2024-12-20:
Kann man eigentlich ein Update auf dem ESP machen oder muss man ihn komplett neu flashen, wenn eine Änderung am Code gemacht wurde?

tomquist, 2024-12-21:
> Kann man eigentlich ein Update auf dem ESP machen oder muss man ihn komplett neu flashen, wenn eine Änderung am Code gemacht wurde?

ESPHome lässt sich per OTA updaten. Das geht AFAIK aber nur über das ESPHome dashboard (z.B. in HomeAssistant) oder über die ESPHome CLI.

tomquist, 2024-12-21:
Es gibt jetzt eine neue Template Version, die nur noch ein minimales Set an Sensoren hat und alle Speicherdaten direkt an MQTT published, ein Topic pro Bluetooth Nachricht. Zum Steuern stehen ein paar separate Topic zur Verfügung. Die ESP-spezifischen Daten werden nach wie vor als ESPHome Sensoren herausgegeben. Hier wäre der Aufwand nicht vertretbar das ganze zusammenzufassen. Ich muss die ganzen Topics noch genauer dokumentieren. Insgesamt sollte das neue Template wesentlich ressourcenschonender sein, für den Preis einer aufwändigeren Integration in die Heim Automatisierung (je nach System). Es gibt auch noch Potenzial das ganze Image noch kleiner zu machen. Werde mir das eventuell anschauen, wenn ich wieder etwas Zeit finde.

tomquist, 2024-12-23:
Topics sind jetzt alle dokumentiert.

closed 2024-12-23 by tomquist