tomquist/esphome-b2500 #14: Workflow erstellt das Image nicht

jmk777jmk, 2024-07-14:
Ich versuche schon seit 2 Tagen über den workflow ein Image zu erstellen aber das klappt leider nicht und scheint immer abgebrochen zu werden.
Leider hab ich keine Ahnung warum.

**Steps to reproduce**

**Expected behavior**

**Actual behavior**

**Additional context**

**Screenshots**

**Device information**

- ESP Board: [e.g. ESP32-C3]

- B2500 Brand: [e.g. Marstek]

- B2500 Version: [ v1, ]

**Configuration**

```json
{
  "name": "speicherverwaltung",
  "friendly_name": "Speicher Verwaltung",
  "mqtt": {
    "topic": "b2500",
    "broker": "192.168.178.66",
    "port": 1883,
    "username": "Sonoff",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "SmartHome24",
    "password": "***"
  },
  "board": "esp32dev",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "storages": [
    {
      "name": "Greensolar Speicher #1",
      "version": 1,
      "mac_address": "***"
    },
    {
      "name": "BluePalm Speicher #2",
      "version": 1,
      "mac_address": "***"
    }
  ],
  "poll_interval_seconds": "30",
  "variant": "auto",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_timer_query": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_dio_flash_mode": false
}
```

tomquist, 2024-07-14:
Das Problem ist das `#` Zeichen im Namen der Speicher. Versuche das mal zu entfernen.

jmk777jmk, 2024-07-14:
Super
Hat geklappt...wieder was dazu gelernt
Dankeschön

Jens

closed 2024-07-20 by tomquist