tomquist/esphome-b2500 #15: Wifi from ESP32 disconnect after a few seconds

r5gt, 2024-07-30:
**After I install the project, the Wifi works for only a few seconds and than stops**

**Device information**

- ESP Board: ESP-WROOM-32

- B2500 Brand: Marstek

- B2500 Version: firmware V2.14

**Configuration**

```json
{
esphome:
  name: "esphome-web-41da98"
  friendly_name: "B2500"
  on_boot:
    priority: -100
    then:
      - lambda: id(b2500_device_generation_1).publish_state("1");
      - lambda: id(b2500_device_name_1).publish_state("B2500");
      #- switch.turn_off: b2500_out1_active_1
      #- switch.turn_off: b2500_out2_active_1
      #- switch.turn_off: b2500_pv2_passthrough_enabled_1
      - delay: 100ms

esp32:
  board: "esp32dev"
  flash_size: 4MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      COMPILER_OPTIMIZATION_SIZE: y
    advanced:
      ignore_efuse_mac_crc: false

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "yyyyyyyyyyyyyyyyyyyyyyyy"

#wifi:
#  ssid: !secret wifi_ssid
#  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Hai-Esp32 Fallback Hotspot"
#    password: "zzzzzzzzzzzzzzzzzzzzzzzz"

#captive_portal:

# Enable logging
logger:
  level: INFO
  tx_buffer_size: 256
  on_message:
    level: WARN
    then:
      lambda: |-
        if (strstr(message, "BT_BTC") != NULL) {
          id(mqtt_client).publish("b2500/debug",message);
          id(mqtt_client).publish("b2500/debug","B_T_C FOUND: restart ESP32");
          id(controller_restart).press();
        }
        if (strstr(message, "btc_transfer_context") != NULL) {
          id(mqtt_client).publish("b2500/debug",message);
          id(mqtt_client).publish("b2500/debug","B_T_C FOUND: restart ESP32");
          id(controller_restart).press();
        }
        if (strstr(message, "mode: single") != NULL) {
          id(mqtt_client).publish("b2500/debug",message);
          id(mqtt_client).publish("b2500/debug","single mode: restart ESP32");
          id(controller_restart).press();
        }
        if (strstr(message, "handle") != NULL) {
          id(b2500_internal_error_count)++;
          char mvalue[48];
          snprintf(mvalue, 48,"handle error counter: %i",id(b2500_internal_error_count));
          id(mqtt_client).publish("b2500/debug",mvalue);
          if(id(b2500_internal_error_count) > 8) {
            ESP_LOGI("b2500","handle error counter: RESET");
            id(b2500_internal_error_count) = 0;
            id(mqtt_client).publish("b2500/debug","handle error counter: RESET");
            id(controller_restart).press();
          }
        }

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 15min
  fast_connect: off
  manual_ip:
    static_ip: "192.168.0.2"
    gateway: "192.168.0.1"
    subnet: "255.255.255.0"
    dns1: "192.168.0.1"
  on_connect:
    - logger.log:
        level: INFO
        format: "Connected to WiFi. Starting BLE scan."
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - logger.log:
        level: INFO
        format: "Disconnected from WiFi. Stopping BLE scan."
    - esp32_ble_tracker.stop_scan:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ESPHome-b2500-wifi"
#web_server:
#  port: 80
#  ota: false
#  js_include: "./v2/www.js"
#  js_url: ""
#  version: 2
captive_portal:

mqtt:
  id: mqtt_client
  #broker: "core-mosquitto"
  #port: 1883
  broker: 172.30.33.0
  username: "homeassistant"
  password: "wwwwwwwwwwwwwwwwwww"
  discovery: False
  reboot_timeout: 0s
  topic_prefix: "b2500"
  log_topic: "b2500/debug"
  on_connect:
    - logger.log:
        level: INFO
        format: "MQTT connected"
  on_disconnect:
    - logger.log:
        level: INFO
        format: "MQTT disconnected"
}
```
[B2500-ESP32-logs.pdf](https://github.com/user-attachments/files/16433534/B2500-ESP32-logs.pdf)

tomquist, 2024-08-04:
This is likely the same issue as in #8. I released a new beta version 2. You can select the version at the very top of the form. Can you try if that fixes the issue for you?

r5gt, 2024-08-04:
Thanks for the suggestion. I tried but it doesn't work.

tomquist, 2024-08-04:
What did you try, and which part didn't work?

r5gt, 2024-08-04:
I tried with version 2 Beta, but it seems to be worse, in fact it never connects. I went back with version 1. For a few seconds the Wifi network works and the sensors of the B2500 are read correctly in Home Assistant. But shortly after the Wifi stop to work. I attach a screenshot of the log after the reset of the device.
[image]

tomquist, 2024-08-04:
I'm not sure what the issue is, and the screenshots aren't very helpful either. Ideally, you'd send some logs in text form. Ideally, those would be logs captured from the serial port at the moment when the ESP32 disconnects from the Wi-Fi. Also, the config you attached above doesn't look like it was generated from the config generator. It looks more like you're using the original config from noone2k.

r5gt, 2024-08-05:
Hello, i have notice that the problem with the wifi reset after a few seconds was a bad parameter in Mqtt configuration ... wrong IP/user/password.
Now the wifi is OK, the web page is OK (all sensor are display correct - see the image attached) but into HA the sensors doesn't appeare. 
[image]
I have an B2500 versione 2 firmware 2.14 (or 214). Tkhs in advance. Best regards.
[logs_esphome-web-41da98_logs (1).txt](https://github.com/user-attachments/files/16503153/logs_esphome-web-41da98_logs.1.txt)
[esphome-web-41da98-yaml.txt](https://github.com/user-attachments/files/16503210/esphome-web-41da98-yaml.txt)

tomquist, 2024-08-05:
You need to enable the checkbox „Discovery“ in the MQTT section to make the device appear in HomeAssistant.

r5gt, 2024-08-05:
Now It works. All sensors are display in HA. Thank you very much.

closed 2024-08-05 by r5gt