tomquist/esphome-b2500 #17: Beta V2 :  Passthrouh wird nicht mehr abgefragt

knickohr, 2024-08-03:
Bei der neuen Beta V2 kann man den Passthrouh  der Speicherversion V1 nicht mehr abfragen und läßt sich über MQTT auch nicht mehr auslesen/setzen. Ist wohl der Rationalisierung zum Opfer gefallen 😲

tomquist, 2024-08-03:
Das geht jetzt über das topic `b2500/1/battery/charge_mode`.

closed 2024-08-03 by tomquist

knickohr, 2024-08-03:
Was wurde noch alles bezüglich MQTT verändert ? 😅

tomquist, 2024-08-03:
Die Zellspannung kommt jetzt als JSON Objekt über das Topic `b2500/1/device/cell_voltage`. Außerdem gibt es kein Topic mehr, das die Summe aller Spannungen über alle Speicher hinweg enthält. Füge ich eventuell bei Bedarf irgendwann noch hinzu. Ansonsten sollte glaube ich soweit alles identisch sein.

knickohr, 2024-08-03:
JSON ist toll, muß ich nur meine Flows in Node-Red anpassen 😉

Also es fehlen 
- die gesamte Summenspannung der Batterie
-  Dann vermutlich auch der errechnete SOC
- Und noch 2 oder 3 Werte aus diesen Branch (min, avg, max. der Zellspannungen)
?

Das esp-Reboot command-topic habe ich auch manuell anlegen müssen (war aber schon vorher nicht drin).

Ich muß mir das in Ruhe anschauen. Der Compiler rennt auf einem Error. Vermutlich muß ich wieder mal das ESPhome updaten.

time_entity.h fehlt

tomquist, 2024-08-03:
Das JSON der Zellspannung sieht so aus:
```json
{
    "cells":[3.195,3.192,3.193,3.192,3.192,3.193,3.192,3.192,3.191,3.192,3.192,3.192,3.166,3.192],
    "min": 3.166,
    "max": 3.195,
    "avg": 3.190428571
}
```

Summe, min, max und avg sind also drin. Den errechneten SOC habe ich weggelassen. Die alte Berechnung war nur eine Approximation, die sehr ungenau war. Wenn ich da was einbaue, dann über eine genauere Approximation. Wer den errechneten SOC in der Zwischenzeit haben will, darf sich selbst eine Heuristik aussuchen und aus den Zellspannungen ermitteln.

Timer entity wurde erst mit ner neueren ESPHome version eingeführt. Du kannst dir aber ein Image direkt über den Config-Generator erzeugen, falls du gerade nicht updaten kannst.

knickohr, 2024-08-03:
Wo ist die Summe über die gesamte Batterie ?  Egal, kann man selbst zusammen rechnen 😉

Ja, der SOC ist Blödsinn. Keine Ahnung warum die das so zusammen gerechnet haben. Oder kommt das sogar gar nicht aus dem Speicher ?

Wenn Du mir noch den esp-Reboot Topic mit einbaust, dann könnte ich tatsächlich das fertige Binary nehmen. Aber laß gut sein, es ist wohl besser das ESPhome upzudaten. Irgendwann fällt mir das wieder auf die Füße.

tomquist, 2024-08-03:
Oh ja stimmt, die Summe fehlt. Kann mich nicht mehr erinnern, warum ich die weggelassen habe. Aus dem Speicher kommt schon ein SoC (ist auch in der neuen config drin unter `b2500/1/battery/remaining_percent`), aber die alte config hatte noch einen aus den Spannungswerten errechneten SoC. Der wurde in der config selbst berechnet.

Den reboot gibt es schon: `b2500/1/reboot/set`

knickohr, 2024-08-04:
Sorry, ich muß Dich etwas quälen.

Habe zwar jetzt das neueste ESPhome drauf, das Problem mit time_entity.h ist gelöst, aber jetzt hagelt es massenhaft Fehler :

ubuntu@Lunix-PC:~/test/config$ ls -la
insgesamt 124
drwxr-xr-x 2 ubuntu uuntu  4096 Aug  4 11:56 .
drwxr-xr-x 3 ubuntu uuntu  4096 Aug  4 11:56 ..
-rw-r--r-- 1 ubuntu uuntu 19111 Aug  4 11:56 config1.yaml
-rw-r--r-- 1 ubuntu uuntu 97680 Aug  4 11:56 config2.yaml
ubuntu@Lunix-PC:~/test/config$ esphome run config1.yaml 
INFO ESPHome 2024.7.3
INFO Reading configuration config1.yaml...
INFO Cloning https://github.com/tomquist/esphome-b2500@main
INFO Fetching main
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing bc2500-1-ble-idf (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/bc2500-1-ble-idf/src/esphome/components/b2500/b2500_base.o
Compiling .pioenvs/bc2500-1-ble-idf/src/esphome/components/b2500/b2500_codec.o
Compiling .pioenvs/bc2500-1-ble-idf/src/esphome/components/b2500/b2500_state.o
Compiling .pioenvs/bc2500-1-ble-idf/src/esphome/components/b2500/b2500_v1.o
Compiling .pioenvs/bc2500-1-ble-idf/src/esphome/components/b2500/b2500_v2.o
Compiling .pioenvs/bc2500-1-ble-idf/src/esphome/components/b2500/binary_sensor/b2500_binary_sensor_base.o
In file included from src/esphome/components/b2500/b2500_v2.cpp:3:
src/esphome/components/b2500/b2500_v2.h:22:44: error: 'datetime' has not been declared
   void set_timer_start_datetime(int timer, datetime::TimeEntity *time) { this->timer_start_[timer] = time; }
                                            ^~~~~~~~
src/esphome/components/b2500/b2500_v2.h:22:65: error: expected ',' or '...' before '*' token
   void set_timer_start_datetime(int timer, datetime::TimeEntity *time) { this->timer_start_[timer] = time; }
                                                                 ^
src/esphome/components/b2500/b2500_v2.h:23:42: error: 'datetime' has not been declared
   void set_timer_end_datetime(int timer, datetime::TimeEntity *time) { this->timer_end_[timer] = time; }
                                          ^~~~~~~~
src/esphome/components/b2500/b2500_v2.h:23:63: error: expected ',' or '...' before '*' token
   void set_timer_end_datetime(int timer, datetime::TimeEntity *time) { this->timer_end_[timer] = time; }
                                                               ^
src/esphome/components/b2500/b2500_v2.h:39:3: error: 'datetime' does not name a type; did you mean 'adjtime'?
   datetime::TimeEntity *timer_start_[3]{nullptr};
   ^~~~~~~~
   adjtime
src/esphome/components/b2500/b2500_v2.h:40:3: error: 'datetime' does not name a type; did you mean 'adjtime'?
   datetime::TimeEntity *timer_end_[3]{nullptr};
   ^~~~~~~~
   adjtime
src/esphome/components/b2500/b2500_v2.h: In member function 'void esphome::b2500::B2500ComponentV2::set_timer_start_datetime(int, int)':
src/esphome/components/b2500/b2500_v2.h:22:80: error: 'class esphome::b2500::B2500ComponentV2' has no member named 'timer_start_'; did you mean 'set_timer_start'?
   void set_timer_start_datetime(int timer, datetime::TimeEntity *time) { this->timer_start_[timer] = time; }
                                                                                ^~~~~~~~~~~~
                                                                                set_timer_start
src/esphome/components/b2500/b2500_v2.h:22:106: error: expected primary-expression before ';' token
   void set_timer_start_datetime(int timer, datetime::TimeEntity *time) { this->timer_start_[timer] = time; }
                                                                                                          ^
[... 19 lines cut ...]
                                                                ^~~~~~~~~~~~
                                                                set_timer_start
src/esphome/components/b2500/b2500_v2.cpp:119:29: error: 'class esphome::b2500::B2500ComponentV2' has no member named 'timer_start_'; did you mean 'set_timer_start'?
           auto call = this->timer_start_[i]->make_call();
                             ^~~~~~~~~~~~
                             set_timer_start
src/esphome/components/b2500/b2500_v2.cpp:125:17: error: 'class esphome::b2500::B2500ComponentV2' has no member named 'timer_end_'; did you mean 'set_timer_end'?
       if (this->timer_end_[i] != nullptr) {
                 ^~~~~~~~~~
                 set_timer_end
src/esphome/components/b2500/b2500_v2.cpp:128:19: error: 'class esphome::b2500::B2500ComponentV2' has no member named 'timer_end_'; did you mean 'set_timer_end'?
[... cut ...]
src/esphome/components/b2500/b2500_v2.cpp:129:29: error: 'class esphome::b2500::B2500ComponentV2' has no member named 'timer_end_'; did you mean 'set_timer_end'?
           auto call = this->timer_end_[i]->make_call();
                             ^~~~~~~~~~
                             set_timer_end
*** [.pioenvs/bc2500-1-ble-idf/src/esphome/components/b2500/b2500_v2.o] Error 1
======================================================================== [FAILED] Took 82.92 seconds ========================================================================
ubuntu@Lunix-PC:~/test/config$

tomquist, 2024-08-04:
Habe gerade nochmal etwas gepushed, das das Problem beheben sollte. Habe bisher immer nur mit einem v1 & v2 gemeinsam getestet, nie mit nur v1.

knickohr, 2024-08-04:
Jepp, es compiliert gerade :+1: 

Bis jetzt ohne Fehler

knickohr, 2024-08-04:
Ok, es lief durch 😎

Aber eine Frage zum MQTT :

Werden jetzt nur noch Daten gesendet wenn sich etwas ändert ?  Ansonsten sieht mir das etwas leer aus 🤔

[image]

Nicht falsch verstehen, es ist gut wenn nur noch was bei Änderung kommt. Ich muß es halt wissen weil ich dann meine Flows anpassen muß.

tomquist, 2024-08-04:
Ja, es werden nur noch Änderung gepushed. Die Idee war, dass dadurch ein paar Ressourcen gespart werden können, sehe aber, dass das evtl. Für die volatilen Werte sollte das ok sein. Die Werte, die sich nicht so häufig ändern werden glaube ich sowieso retained, feshalb das in der Praxis kein Problem sein sollte. Lasse mich aber gerne überzeugen den ein oder anderen Wert häufiger zu senden, falls das notwendig sein sollte.

knickohr, 2024-08-04:
Nein, das ist perfekt. Nur Änderungen senden. Und über diese retained Werte müssen wir uns auch nochmal unterhalten. Meiner Meinung nach bedarf es nur bei sehr wenigen Werten eines retained Flags.

Einzig den last_response würde ich immer senden wenn der Speicher abgefragt wurde und auch geantwortet hat, egal ob sich die Werte änderten oder nicht.

tomquist, 2024-08-04:
Ja, das Thema `retain` habe ich mir noch nicht angeschaut. ESPHome retained standardmäßig alle topics. Wenn man das nicht will, muss man es für jede Komponente einzeln ausschalten. Welche Topics denkst du denn sollten nicht retained werden? Vermutlich alle, die sich häufig ändern.

knickohr, 2024-08-04:
Korrekt, alle die sich häufig ändern und bereits nach wenigen Sekunden keine Aussage mehr haben.

Hier habe ich mal bei noone2k schon was dazu geschrieben. Müßte man nochmal drüber schauen da sich einiges geändert hat.

https://github.com/noone2k/hm2500pub/issues/8

tomquist, 2024-08-04:
Habe mal `retain` für ein paar topics abgeschaltet.

knickohr, 2024-08-06:
Ja, und ein Energy-Meter eingebaut 😎