tomquist/esphome-b2500 #19: Not wokring on ESP32-H2

hanno1seibod, 2024-08-12:
Wird im Config-Generator unter "variant" H2 ausgewählt, so schläft der Workflow fehl.
Wird im Config-Generator unter "variant" auto ausgewählt, so hängt sich der ESP nach dem Flashen auf.

**Device information**

- ESP Board: ESP32-H2

- B2500 Brand: Green Solar

- B2500 Version: v3

**Configuration**

```json
{
  "template_version": "v2",
  "name": "b2500",
  "friendly_name": "B2500",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "topic": "b2500",
    "broker": "192.168.10.28",
    "port": 1883,
    "username": "---",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "---",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "auto",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "---",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": false,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": false,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "B2500",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2024-08-12:
Das Board wir wohl aktuell von ESPHome nicht unterstützt, aber du kannst mal versuchen bei Board `esp32-c3-devkitm-1` und dann bei Variant `esp32h2` anzugeben. Das soll laut [diesem issue](https://github.com/esphome/issues/issues/5032) funktionieren.

noone2k, 2024-08-12:
ohne gegentest:

ab esp-idf 5.3 wird der h2 direkt unterstützt.
ob die platformio bereits den layer unterstützt, no idea ...

ein versuch wäre es wert ( habe bei mir lokal meistens die neuestens versionen aktiviert )
f.ex.:

```
esp32:
  #board: az-delivery-devkit-v4
  board: esp32dev
  framework:
    #platform_version: 6.7.0
    platform_version: 6.8.1
    #version: 5.2.2
    version: 5.3.0
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true

```