tomquist/esphome-b2500 #21: Script "b2500_script_set_timer_enable" fehlt in v2

DjLex2021, 2024-09-05:
Hi @tomquist,

ich habe ein Problem beim Build von der v2.
Es wird auf das Script "b2500_script_set_timer_enable" verwiesen, welches aber nicht existiert.
Daher schlägt der Build leider fehl.

Ist das für die fest eingestellten Entladezeiten gedacht, wenn kein SmartMeter verwendet wird?
Dann würde ich die 7 Switche einfach entfernen und erneut prüfen.

[image]

tomquist, 2024-09-05:
Kannst du hier bitte die config posten, die du beim drücken auf dem „Report an issue“ Button bekommst?

DjLex2021, 2024-09-05:
„Report an issue“ Button?

Wenn ich es in ESPHome baue kommt der Fehler oben im Screenshot

Wenn ich es über Git baue kommt das hier:
https://github.com/tomquist/esphome-b2500/actions/runs/10720453182/job/29726736774

DjLex2021, 2024-09-05:
Sorry!! Ich war blind ... ganz oben !!

```json
{
  "template_version": "v1",
  "name": "b2500-node",
  "friendly_name": "B2500 Node",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "topic": "b2500-node",
    "broker": "192.168.2.60",
    "port": 1883,
    "username": "mqttuser",
    "password": "***",
    "discovery": true
  },
  "wifi": {
    "ssid": "Appalachia",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "auto",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": false,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "B2500",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2024-09-05:
Ich muss mir das Problem später genauer anschauen, aber in der Zwischenzeit kannst du unter Advanced den Schalter bei „Enable Timerquery“ anschalten. Das sollte das Problem beheben.

DjLex2021, 2024-09-06:
Hi @tomquist 
ich verwende nun die HW v1. Die funktioniert erstmal auch.

Die v2 bekomme ich nicht zum laufen.
Er stellt ewig keine Verbindung zum WLAN her und nach ner Zeit kommt folgende Meldung

`[I][esp-idf:000][sys_evt]: I (3674) esp_netif_handlers: sta ip: 192.168.2.89, mask: 255.255.255.0, gw: 192.168.2.1

[I][wifi:617]: WiFi Connected!
[W][component:157]: Component mqtt set Warning flag: unspecified
[I][mqtt:244]: Connecting to MQTT...
[I][main:1049]: Connected to WiFi. Starting BLE scan.
[W][component:170]: Component wifi cleared Warning flag
[I][esp-idf:000][wifi]: W (7338) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (7440) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (7542) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (7647) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (7747) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (7849) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (8054) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (8157) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (8259) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (8367) wifi:
[I][esp-idf:000][wifi]: mem fail
[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (8460) wifi:
[I][esp-idf:000][wifi]: m f null

[I][esp-idf:000][wifi]: 

[I][esp-idf:000][wifi]: W (12365) wifi:
[I][esp-idf:000][wifi]: mem fail
[I][esp-idf:000][wifi]: 

[I][main:1165]: MQTT connected
[W][component:170]: Component mqtt cleared Warning flag
[I][mqtt:284]: MQTT Connected!

[13:10:09]abort() was called at PC 0x401c6226 on core 0
[13:10:09]
[13:10:09]
[13:10:09]Backtrace: 0x4008247e:0x3fff0b20 0x40091495:0x3fff0b40 0x400970d2:0x3fff0b60 0x401c6226:0x3fff0bd0 0x401c4dd7:0x3fff0bf0 0x401c4d3e:0x3fff0c10 0x400d61a5:0x3fff0c30 0x40101c9d:0x3fff0c50 0x40102655:0x3fff0c70 0x4012b82a:0x3fff0c90 0x4012b7c7:0x3fff0cb0
[13:10:09]
[13:10:09]
[13:10:09]
[13:10:09]
[13:10:09]ELF file SHA256: 428404b786d73345
[13:10:09]
[13:10:09]Rebooting...`

tomquist, 2024-09-11:
Das ursprüngliche Problem von oben sollte jetzt behoben sein. Wegen des WLAN Problems, hast du schonmal die configuration version v2 getestet?

DjLex2021, 2024-09-12:
Ob jetzt direkt das WLAn das Problem ist, möcht ich nichtmal sagen. Es kommt bei der v2 Hardware und v1 Config jedenfalls zum Buffer overflow. Mit der v2 Config läuft es aktuell sehr gut. Danke hierfür!

Auch wenn etwas offtopic, aber kannst du mir sagen, wo die Daten vom Sensor "Total Daily Energy Out" herkommen? Stellt den der B2500 bereit oder kommet der vom Script? Dieser Senor macht sehr seltsame Sprünge. Immer dann, wenn eine Zeitlang keine Energie abgegeben wurde.

[image]

tomquist, 2024-09-12:
Der kommt von Script. Das ist ein Sensor der über die Ausgangsleistung integriert. Bist du dir sicher, dass in den Zeiten keine Energie abgegeben wurde? Sieht für mich eher so aus als wäre in der Zeit die Kommunikation zwischen ESP32 und MQTT Broker unterbrochen gewesen. Die Sprünge nach unten machen Sinn. Am Ende des Tages wird der Sensor auf 0 zurückgesetzt.

DjLex2021, 2024-09-12:
Ne da wurde keine Energie abgegeben. Das deckt sich auch mit der Zeit, in der der Speicher geladen wurde.
Ausserdem könne bei 2,24 kWh Speicher keine 5kWh abgegeben werden. Ich habe zwei dieser Speicher und die machen beide diese Sprünge. Auch teilweise über 7 kWh.

Zum Tageswechsel passt es wunderbar. die Kurve steigt langsam nach oben, dann wird der Speicher geladen und ich habe eine gerade Linie. Pass also auch. Dann jedoch ist die Sonne weg und der Speicher speist ein. Immer dann gibt es diesen Sprung nach oben

Ich komm einfach nicht dahinter, was hier das Problem ist...

DjLex2021, 2024-09-12:
Ich habe dir hier mal beide Sensoren übereinander gelegt, damit du den Zusammenhang besser sehen kannst.
Am 09.09. habe ich Nachmittag testweise einmal auf laden & entladen gestellt. Daher geht die Ladekurve weiter hoch.

Aber du kannst erkennen, dass der Sprung immer mit start der Einspeisung stattfindet

[image]

tomquist, 2024-09-12:
Oh, ich glaube ich weiß was das Problem ist. Es wird standardmäßig nach rechts integriert. Es wird angenommen, dass die Leistung konstant bleibt und dem späteren Messwert entspricht. Das stimmt aber natürlich nicht. Wenn keine Energie abgegeben wird, nimmt das System dann den ersten Wert für den gesamten Zeitraum an, der bei Energieabgabe gemessen wird. Lösung ist das ganze nach links zu aggregieren. Ich fixe das später.

tomquist, 2024-09-13:
Habe eben ein update gepushed. Es wird jetzt nach links integriert.

closed 2024-09-13 by tomquist