tomquist/esphome-b2500 #25: IO-Broker Steuerbefehle

joewest74, 2024-10-09:
Moin,

erstmal Danke, lief so schön Easy mit dem Tool. Esp läuft, auslesen kann ich alles, alles sauber in den Objekten vom IO-Broker angelegt.

Einfacher geht es kaum.

Wo es bei mir aber total harkt ist den Akku zu Steuern, über das Webinterface des ESP geht es natürlich sofort. Würde Ihn aber gerne per NoteRed oder Script steuern. Da fehlt mir der richtige Ansatz. Was ich finde ist per CD20 Befehl, aber gilt das nicht nur wenn ich direkt per Mqtt mit dem Akku spreche, also ohne ESP?

Kannst mir da mal auf die sprünge helfen? Oder sende ich auch einfach zb. cd=20,md=0,a1=1,b1=0:30,e1=6:30,v1=260,a2=0,b2=12:30,e2=20:30,v2=123,a3=0,b3=12:30,e3=20:30,v3=123 
an den Esp? 

Tausend Dank für Deine Arbeit und das veröffentlichen.

Jörn

HelmutFi, 2024-10-16:
Das würde mich auch interessieren. Ich verwende im Backend IPSymcon und habe so ziemlich alle Topics im System. Leider hab ich den Akku noch nicht (ist am Weg), daher nur eine Trockenübung. Auch aus IPS kann ich änderbare Werte über MQTT zum ESP schicken. Nur tut sich da nichts. Zum Beispiel Bluetooth ein/aus. In IPS wird mir eine String Variable anglegt mit Wert "OFF". Wenn ich das testweise auf "ON" ändere, ändert sich auf der anderen Seite nichts.

tomquist, 2024-12-20:
[Hier](https://github.com/tomquist/esphome-b2500#mqtt-topics) gibt es jetzt eine ausführliche Dokumentations der Topics.

commit 9bb1d5a8 (2024-12-20): Add documentation for MQTT topics

Fixes #25
commit d15d534d (2024-12-20): Add documentation for MQTT topics (#37)

Fixes #25
closed 2024-12-20 by tomquist