tomquist/esphome-b2500 #27: fehlende MQTT Items bei Configuration Version 2

Gebegern, 2024-10-19:
Vorab: Klasse Arbeit!!
was mir aufgefallen ist:
wird der yaml Code mit der Configuration V2 erzeugt, kann ich im MQTT Explorer nur 3 Items im battery Folder sehen. 
wir der yaml Code mit der Configuration V1 erzeugt erscheinen 5  Items. 
In der V2 Version fehlen leider die Einträge "remaining_percent" und "remaining_capacity".
siehe Screenshot

**Steps to reproduce**

**Expected behavior**

**Actual behavior**

**Additional context**

**Screenshots
[image]
**

**Device information**

- ESP Board: [ESP32-C3]

- B2500 Brand: [Marstek]

- B2500 Version: [v2]

**Configuration**

```json
{
  "template_version": "v1",
  "name": "akku-b2500",
  "friendly_name": "Akku B2500",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "topic": "akku-b2500",
    "broker": "192.168.178.61",
    "port": 1883,
    "username": "mqtt-user",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "Gluecksburg",
    "password": "***"
  },
  "board": "az-delivery-devkit-v4",
  "variant": "auto",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "Akku-B2500 Fallback Hotspot",
    "enable_captive_portal": true
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "Akku-B2500-1",
      "version": 2,
      "mac_address": "***"
    },
    {
      "name": "Akku-B2500-2",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

ugath, 2024-10-20:
Einfach etwas abwarten!
Die Werte kommen irgendwann (wahrscheinlich wenn sich im Speicher was ändert).
[image]

MtheK, 2025-06-08:
Liegt es am Speicher oder am esphome dass Werte nur bei Änderungen geschickt werden?

Oft fehlen Werte sehr lang, zB habe ich vor 3h esphome neu gestartet, in den Marstek App den DOD mehrfach geändert, sehe den DOD aber immer noch nicht im MQTT:

[image]

Kann man einstellen dass alle Werte regelmäßig verschickt werden egal ob sie sich geändert haben?

tomquist, 2026-03-15:
Native API should solve this