tomquist/esphome-b2500 #33: Compiling error

gaugstein, 2024-12-18:
<pre>
src/esphome/components/b2500/b2500_base.cpp:288:43: note: format string is defined here
  288 |   ESP_LOGCONFIG(TAG, "  Update Interval: %d ms", this->update_interval_);
      |                                          ~^
      |                                           |
      |                                           int
      |                                          %ld
</pre>

tomquist, 2024-12-18:
Bei welcher config und wie genau baust du das Image?

tomquist, 2024-12-18:
Probier es nochmal mit der neusten Version

gaugstein, 2024-12-18:
Successfully created esp32c3 image.

Dafür geht jetzt der 32s nicht mehr:

<pre>
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/api_connection.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/api_server.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/list_entities.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/proto.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/api/user_services.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/b2500/b2500_base.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/b2500/b2500_codec.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/b2500/b2500_state.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/b2500/b2500_v1.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/b2500/b2500_v2.o
Compiling .pioenvs/b2500b-ble-idf/src/esphome/components/b2500/binary_sensor/b2500_binary_sensor_base.o
src/esphome/components/b2500/b2500_codec.cpp: In member function 'bool esphome::b2500::B2500Codec::parse_timer_info(uint8_t*, uint16_t, esphome::b2500::TimerInfoPacket&)':
src/esphome/components/b2500/b2500_codec.cpp:220:3: error: expected ',' or ';' before 'if'
   if (data_len < header_size + payload_size) {
   ^~
*** [.pioenvs/b2500b-ble-idf/src/esphome/components/b2500/b2500_codec.o] Error 1
</pre>

tomquist, 2024-12-18:
Probier nochmal.

gaugstein, 2024-12-18:
Test läuft, aber das fällt schon mal auf:
<pre>
src/esphome/components/sntp/sntp_component.cpp: In member function 'virtual void esphome::sntp::SNTPComponent::setup()':
src/esphome/components/sntp/sntp_component.cpp:34:21: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
   34 |   sntp_setservername(0, strdup(this->server_1_.c_str()));
      |   ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/esphome/components/sntp/sntp_component.cpp:5:
/config/.esphome/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:263:6: note: declared here
  263 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:36:23: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
   36 |     sntp_setservername(1, strdup(this->server_2_.c_str()));
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/config/.esphome/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:263:6: note: declared here
  263 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:39:23: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
   39 |     sntp_setservername(2, strdup(this->server_3_.c_str()));
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/config/.esphome/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:263:6: note: declared here
  263 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
src/esphome/components/sntp/sntp_component.cpp:45:12: warning: 'void sntp_init()' is deprecated: use esp_sntp_init() instead [-Wdeprecated-declarations]
   45 |   sntp_init();
      |   ~~~~~~~~~^~
/config/.esphome/platformio/packages/framework-espidf/components/lwip/include/apps/esp_sntp.h:269:6: note: declared here
  269 | void sntp_init(void)
</pre>

gaugstein, 2024-12-18:
Der erste lief schonmal durch:
<pre>
Successfully created esp32c3 image.
Linking .pioenvs/b2500b-ble-idf/firmware.elf
RAM:   [==        ]  16.6% (used 54296 bytes from 327680 bytes)
Flash: [==========]  99.5% (used 1956386 bytes from 1966080 bytes)
</pre>
Teste noch den 32s ..

gaugstein, 2024-12-18:
Lief auch gut, jetzt warten wir mal auf neuerlichen Frost ;)
<pre>
Successfully created esp32 image.
Linking .pioenvs/b2500b-ble-idf/firmware.elf
RAM:   [==        ]  17.8% (used 58232 bytes from 327680 bytes)
Flash: [========= ]  92.6% (used 1698485 bytes from 1835008 bytes)
Building .pioenvs/b2500b-ble-idf/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
</pre>

closed 2024-12-19 by tomquist