tomquist/esphome-b2500 #46: Battery info doesn't update

thejokoono, 2024-12-23:
The ESP only sends battery data once, then stops sending MQTT info about the batteries. Debug info is sent constantly without issue. See timestamps of mqtt messages:
[image]

Flash build to ESP32, plug in ESP32 close to battery, then monitor via MQTT explorer

MQTT explorer should receive updates in 5s intervals about the battery

updates are only sent once when the ESP32 boots

**Screenshots**

**Device information**

- ESP Board: APKLVSR ESP32 ESP-WROOM-32 NodeMCU Lua Development Board

- B2500 Brand: BluePalm

- B2500 Version: v1

- Build: [angry-zany-koala-840] 

**Configuration**

```json
{
  "template_version": "v1",
  "name": "b2500ii",
  "friendly_name": "B2500ii",
  "poll_interval_seconds": "5",
  "log_level": "DEBUG",
  "mqtt": {
    "topic": "solar/akku",
    "broker": "192.168.2.223",
    "port": 1883,
    "username": "Mosquitto",
    "password": "***",
    "discovery": true
  },
  "wifi": {
    "ssid": "***",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "esp32",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": true,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "Akku1",
      "version": 1,
      "mac_address": "***"
    },
    {
      "name": "Akku2",
      "version": 1,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2024-12-23:
Looks like you're using the "Legacy" configuration version right now. I won't maintain this version anymore. Can you test with the "Native ESPHome component" version again?

tomquist, 2024-12-23:
Did the SoC change in the timeframe you tested it?

thejokoono, 2024-12-23:
Not by much. 

It seems to be working now, for some reason. Perhaps it was the software update from V135 to V142 that fixed it, but can't say for certain. 

Thanks for your help and the work you do on this project!

closed 2024-12-24 by thejokoono