tomquist/esphome-b2500 #52: Marstek Jupiter Speicher

mark82436, 2024-12-25:
Hallo zusammen,

eine Frage ist es evt. angedacht das Projekt auch für den Marstek Jupiter zu erweiter bzw. ein neues Projekt dafür öffnen?
Es wäre schön, auch diesen Speicher über MQTT in HomeAssitant einzubinden.

Gruß

tomquist, 2024-12-26:
Prinzipiell spricht nichts dagegen. Allerdings besitze ich keinen Jupiter und habe auch nicht vor mir einen zu kaufen. Es müsste also entweder von einem anderen Besitzer umgesetzt werden, oder Marstek stellt mir einen Jupiter kostenlos zur Verfügung (was eher unwahrscheinlich ist 😉).

closed 2024-12-26 by mark82436

damanti-me, 2025-04-22:
Ich würde wirklich gerne bei der Umsetzung helfen, da ich auch einen Jupiter im Zulauf habe. Im Prinzip müssten wir doch "nur" die entsprechenden modbus register neu zuordnen, oder? Die Zuordnung des B2500 wurde beim Marstek support angefragt  nehme ich an. Wie würdest du in groben Zügen vorgehen?