tomquist/esphome-b2500 #56: Threshold lässt sich nicht einstellen

Kawakasi, 2024-12-29:
Ich entschuldige mich schonmal vorab, aber ich bin noch neu in dieser ganzen Thematik und noch auf Hilfe angewiesen.
Folgendes Szenario ist zur Zeit bei mir vorhanden:
Mit Hilfe des B2500 Config Generator ein Image für Version 1 erstellt.
Eine alte opendtu entsprechend umgeflasht. 
Einbindung in HA.
Soweit alles gut und läuft auch.
Ich bekomme es aber leider nicht hin die Entladung zu steuern. Weder über HA direkt, noch über die Web-Oberfläche des ESP.
Der Wert lässt sich zwar einstellen, wird aber nicht übernommen und nach wenigen Sekunden wieder auf 0 gesetzt.
Des weiteren scheint die Verbindung ständig getrennt zu werden wenn ich die Log's richtig interpretiere.
Evtl. kann mir jemand Hilfestellung bieten.
Anbei ein Auszug des Log's.

 16:39:17
[W]
[component:237]
Component esp32_ble took a long time for an operation (103 ms).
16:39:17
[W]
[component:238]
Components should block for at most 30 ms.
16:39:20
[W]
[ble_sensor:078]
Error reading char at handle 21, status=2
16:39:22
[I]
[main:700]
Polling B2500 devices...
16:39:22
[I]
[main:1004]
Performing runtime query for device 1
16:39:22
[I]
[main:1652]
Sending command to device 1: [3] 1
16:39:22
[I]
[main:1136]
Parsing runtime data from device 1
16:39:22
[W]
[component:237]
Component esp32_ble took a long time for an operation (98 ms).
16:39:22
[W]
[component:238]
Components should block for at most 30 ms.
16:39:27
[I]
[main:700]
Polling B2500 devices...
16:39:27
[I]
[main:1004]
Performing runtime query for device 1
16:39:27
[I]
[main:1652]
Sending command to device 1: [3] 1
16:39:27
[I]
[main:1136]
Parsing runtime data from device 1
16:39:27
[W]
[component:237]
Component esp32_ble took a long time for an operation (104 ms).
16:39:27
[W]
[component:238]
Components should block for at most 30 ms.
16:39:32
[I]
[main:700]
Polling B2500 devices...
16:39:32
[I]
[main:1004]
Performing runtime query for device 1
16:39:32
[I]
[main:1652]
Sending command to device 1: [3] 1
16:39:32
[I]
[main:1136]
Parsing runtime data from device 1
16:39:32
[W]
[component:237]
Component esp32_ble took a long time for an operation (93 ms).
16:39:32
[W]
[component:238]
Components should block for at most 30 ms.

Kawakasi, 2024-12-30:
Hat sich erledigt, war ein Konfigurationsproblem.

closed 2024-12-30 by Kawakasi

kduchrow, 2025-02-13:
Hi @Kawakasi,

Ich habe das gleich problem. Was genau ist denn das "Konfigurationsproblem" ? 

Vielen Dank!

kduchrow, 2025-02-13:
Okay, jetzt geht es bei mir auch. 
Ich hatte die Falsche Version des Speichers ausgewählt. 
Umgestellt auf Version 2 und nun funktioniert es.