tomquist/esphome-b2500 #61: Iobroker MQQT

Tobiasp1983, 2025-01-04:
Hy,
erstmal danke top Arbeit.
Ich habe die Software auf einen ESP geschrieben aber ich bekomme keine Verbindung zusammen,
Im IOBroker kommt:
Cannot configure secure web server, because no certificates found: , ,

tomquist, 2025-01-04:
Ich bräuchte noch ein paar mehr Informationen um zu helfen. Was genau ist das Problem? Wo bzw. in welchem Zusammenhang kommt diese Fehlermeldung genau? Kannst du auf den ESP32 per webinterface zugreifen? Mit welcher config hast du den ESP32 bespielt (am Besten im Config Generator auf „Export as JSON“ drücken, Passwörter entfernen und den Inhalte hier reinkopieren).

Tobiasp1983, 2025-01-04:
Hi, also mittlerweile hab ich es geschafft das der ESP mit dem WLan verbunden ist und mit MQQT vom Iobroker. Auf das webinterface komm ich auch. 
Auch hab ich etliches im Broker zu sehen aber keine Werte von den Akkus. Ich sehe auch den "ctrl" vom Akku. Aber wo sehe ich die Werte selbst?

tomquist, 2025-01-04:
Ich glaube du vermischt da etwas. Das ctrl topic kommt vom Speicher selbst, nicht von ESP. Du kannst den Speicher entweder über den ESP32 integrieren, der sich per Bluetooth die Daten von Speicher holt, oder indem du den Speicher auf deinen eigenen MQTT umkonfigurierst. Bei zweiterem siehst du das ctrl topic im MQTT. Bei ersterem siehst du einzelne Topics für jeden Sensor. Sieht aber so aus als würde sich der ESP32 nicht per Bluetooth mit deinem Speicher verbinden. Ist der ESP32 in Bluetooth-Reichweite? Was sagen die Logs im Webinterface? Zeigt das Webinterface an, dass der Speicher per Bluetooth verbunden ist?

Tobiasp1983, 2025-01-05:
Guten Morgen, 
ah verstehe, ja ist in Reichweite. Ich will die APP und deine Sw nutzen im Ioboker. Reichweite vom Bluetooth ist es ja.
Aber man siehr ned wirklich vel auf der Weboberfläche
[image]

Tobiasp1983, 2025-01-05:
so wenn ich die App öffne, erscheint es auch auf der Weboberfläche
[image]

Woher weiß ich das er mit BT verbunden ist?

tomquist, 2025-01-05:
Ah, sieht so aus als hättest du die „MQTT Relay“ Version ausgewählt im Config Generator ausgewählt. Die macht was anderes.

Siehst du, daher wäre es wichtig gewesen, hier die Config anzuhängen nach der ich oben gefragt habe, das macht es wesentlich einfacher zu helfen 😉

Ändere mal das die Config Version auf „Native ESPHome Component“.

Tobiasp1983, 2025-01-05:
Ah stimmt mein Fehler wollte ich mitsenden. 
Ist die Native Version auch das man beides nutzen kann? Also App und MQQT Homeserver?
Vielen Dankf fpr die Unterstütrzung auf Jedenfall

tomquist, 2025-01-05:
Ich glaube es gibt noch ein paar Verständnisprobleme. Hier mal zusammengefasst deine Optionen:
1. Über die App den Speicher auf deinen eigenen MQTT Broker konfigurieren: Damit sendet der Speicher seine Daten direkt an deinen lokalen MQTT Broker. Die Daten musst du dann selbst über eine Automation anfragen und auswerten, dafür gibt es noch keine Tools, die das einfacher machen. Im Grunde heißt das, die Daten zu parsen und dann in Entitäten zu schreiben und es ist ein wenig Know-How in deiner Heimautomatisierung notwendig. Ein weiterer Nachteil ist, dass dann die App von Remote nicht mehr funktioniert. Dafür gibt es im Config Generator die „MQTT Relay“ Version, die es dann wieder ermöglicht die App zu nutzen. Sie hilft aber (aktuell) nicht dabei die Daten auszuwerten sondern leitet nur die Rohdaten zwischen deinem lokalen MQTT und der Cloud weiter.
2. Den Speicher nicht auf deinen eigenen MQTT konfigurieren und dafür einen ESP32 mit einer Version bespielen, die sich mit dem Speicher per Bluetooth verbindet. Dann kann sich die App zwar nicht mehr per Bluetooth mit dem Speicher verbinden, so lange der ESP läuft, aber dafür. bekommst du den die Daten fertig ausgewertet und kannst es relativ einfach in die Heimautomatisierung integrieren.

Es scheint als ob du aktuell auf dem Weg bist Option 1 umzusetzen, aber Option 2 wäre besser geeignet für dich.

Tobiasp1983, 2025-01-06:
Hi, super funktioniert alles einwandfrei. Danke für die Hilfe

tomquist, 2025-01-06:
Super!

closed 2025-01-06 by tomquist