tomquist/esphome-b2500 #67: esp32-s3-devkitc-1 N16R8 integration with PSRAM

gaugstein, 2025-01-12:
Hey Tom, ich hab mir bei Ali zwei dieser Dev-Boards geholt. Auf einem fehlte für die RGB die besagte Brücke, der andere war zumindest komplett funktionsfähig. Was allerdings bei beiden schief ging war die Verwendung des erweiterten PSRAM. Zum Vergleich hab ich ein Micropython installiert und da fiel mir auf, dass es für genau diese Variante (N16R8) eine eigene Firmware gibt. Wenn du also diese Boards empfiehlst sollte auch folgender Parameter nicht fehlen, sonst funktioniert der PSRAM ned:
<pre>
psram:
  mode: octal
  speed: 80MHz
</pre>

tomquist, 2025-01-12:
Interessant. Soweit ich gelesen habe ist das aber nur wirklich relevant wenn ein Display angeschlossen ist. Nutze das board selbst ohne PSRAM aktiviert zu haben. Oder gibt es sonst irgendwelche Nachteile?

gaugstein, 2025-01-12:
Ich denke du verwendest dann halt nur die 4/8MB und läßt den Rest links liegen. Und ich hab ein paar statements gefunden die explizit die Verwendung von PSRAM unterbinden, d.h. er wird standardmäßig verwendet, insofern er eingebunden ist. Das Modul mit Display hab ich leider geschrottet, aber wenn ein neues da ist werd ich damit einige weitere Tests machen.

tomquist, 2025-01-12:
Habe den PSRAM mal eben bei mir aktiviert. Kann erstmal keinen Unterschied feststellen. Lief vorher schon sehr gut, mit sehr wenigen Neustarts über Tage hinweg. Mal beobachten wie es mit aktiviertem PSRAM aussieht...

gaugstein, 2025-01-12:
Aktuell kämpfe ich noch mit der RGB LED die bei meinen Boards drauf ist. Ich bekomme die nicht runtergeregelt per default. In HomeAssist kann ich sie einstellen, aber als Status für WLAN/MQTT/BLE ist die einfach zu hell. Bei 20-30% wäre sie perfekt.

gaugstein, 2025-01-12:
Beim Thema PSRAM hast du wohl recht, siehe: https://esphome.io/components/psram.html . In deinem Fall wird er offensichtlich nicht verwendet. Hab mal ein paar Sensoren hinzugefügt:
<pre>
sensor:
  - platform: debug
    free:
      name: "${sfx}: Heap Free"
      id: b2500_device_heak_free
      state_topic: "b2500/1/device/heap_free"
#    fragmentation:
#      name: "Heap Fragmentation"
    block:
      name: "${sfx}: Heap Max Block"
      id: b2500_device_heap_max_block
      state_topic: "b2500/1/device/heap_max_block"
    loop_time:
      name: "${sfx}: Loop Time"
      id: b2500_device_loop_time
      state_topic: "b2500/1/device/device_loop_time"
    psram:
      name: "${sfx}: Free PSRAM"
      id: b2500_device_free_psram
      state_topic: "b2500/1/device/free_psram"
</pre>

Und der PSRAM Wert bleibt konstant.