tomquist/esphome-b2500 #68: v2/www.js

gaugstein, 2025-01-13:
Hi Tom, wie kommt eigentlich die www.js zu Stande. Frage deshalb weil mich stört, dass das obere Symbolfenster nur erscheint wenn der Name einem Schema folgt: "B2500 - 1 - XXX". Kann ma das evtl ändern, ich finde die Stelle mit der RegExp nicht mehr und würde das gern durch ein simples B2500 -1 ohne den ganzen Quatsch dahinter ersetzen. In einem normalen Browser schneidet der lange Suffix die wichtigen Infos leider ab:
[image]

tomquist, 2025-01-13:
Der www Code ist hier: https://github.com/tomquist/esphome-webserver-b2500

gaugstein, 2025-01-17:
Ich kann die www.js erzeugen, wobei die V3 aussieht wie die V2 im Ergebnis. Und ich weiß noch nicht wie das mit dem Proxy funktioniert. Jede Änderung auf den Chip zu brennen dauert einfach zu lange.

tomquist, 2025-01-28:
Bist du sicher, dass du eine v3 erzeugt hast? Zum lokalen entwickeln am besten [diese Zeile](https://github.com/tomquist/esphome-webserver-b2500/blob/c4649b8a9bf7cb199a789e3b1bd465a6afae8ba9/packages/v3/vite.config.ts#L12) auf die IP deines ESP32 ändern und dann kann man damit auch lokal arbeiten, ohne jede Änderung auf den Flash zu schreiben.