tomquist/esphome-b2500 #92: Veränderte Entladezeiten in der App V1.6.20

RALFISBORN, 2025-03-12:
Hallo,

mir ist nach dem neusten Update der App Power Zero aufgefallen, dass es nicht mehr fünf separate Timer gibt, sondern jetzt manuell Zeiten hinzugefügt werden können:

[image]

Seit dem ist mir auch aufgefallen, dass ich in HA öfters mal die Meldung im Log habe, dass der Timer 5 Wert außerhalb des gültigen Bereichs liegt (angeblich unter 80W) obwohl dieser, wie die anderen Werte auch, fest auf 80W geschrieben wird. Ich habe nur einen Timer aktiv, über den ich den Ausgabesollwert via Node-Red => MQTT vorgebe.

Ich verwende folgende Versionen:

[image]

Da ich im Grunde eh via Node-Red => MQTT die Ausgangsleistung steuere, würde mir auch ein Timer völlig ausreichen. In der Vergangenheit war es nur so, dass wenn ich die anderen Timer nicht auch default z.B. disabled habe, es ab und an vor kam, dass plötzlich ein weiterer Timer aktiv war. Das war auch unabhängig davon, ob der ESP neu gestartet wurde oder nicht.

Wird das in einer neuen Version berücksichtigt, diese Entladetimer Thematik anzupassen bzw. gibt es dazu konkrete Pläne?

Ansonsten klappt bisher die Entladesteuerung soweit, bis auf die Fehlermeldung im Log.
An der Stelle nochmals vielen Dank für die super Arbeit ;)

Vielen Dank

RALFISBORN, 2025-04-30:
Hat dies schon jemand anders alternativ lösen können?

Dropsmasta, 2025-05-16:
Hi, 

habe ich es richtig verstanden, das du einen Timer von 0-24Uhr hast und bei dem dann die Leistung steuerst?

Wie dynamisch/häufig änderst du hier die Leistung?
Ich hatte die Erfahrung gemacht, das es mehrere Sekunden (nicht gemessen) dauert, bis der Akku die Leistung auch ausgibt und dann kam noch das Delay vom Wechselrichter dazu, so das ich den Gedanken an eine dynamische Steuerung direkt wieder verworfen hatte.

Könntest du vielleicht mal Plots dazu zeigen? 🤔

RALFISBORN, 2025-05-17:
Hey,

ja genau so mache ich es. Ich habe den Timer 3 immer aktiviert und die anderen auf deaktiviert. Und über Node-Red ermittele ich jede Minute den Durchschnittsverbrauch, errechne mir dann was ich vom Speicher benötige und schick dies dann via mqtt als Ausgabeleistung für den Timer 3 raus. Das klappt soweit ganz gut. Die paar Sekunden delay kann ich hier vollends verkraften ;). Ich habe eher die Erfahrung gemacht, dass wenn ich jede Sekunde den Speicherausgang regeln möchte, es eher zu Probleme kommt (z.B. ESP hängt sich auf und wird dann neugestartet).

Wie machst du das dann?

tomquist, 2025-05-17:
Kleiner Hinweis: zu häufiges Schreiben macht langfristig den Flash im Speicher kaputt. Der hat nur eine begrenzte Anzahl an Schreibzyklen. Leider gibt es nur über MQTT Commands die nicht in den Flash geschrieben werden. Über Bluetooth wird immer auf den Flash geschrieben.

RALFISBORN, 2025-05-17:
@tomquist Oh je, das war mir so gar nicht bewusst :(. Der größte Fehlkauf dieser Speicher, echt Wahnsinn. Was ich damit schon Ärger hatte und funktionell frage ich mich sowieso, was sich so Entwickler bzw. Produktmanager dabei denken, so etwas auf den Markt zu bringen. Warum nicht einfach eine lokale MQTT Anbindung direkt onBoard und das ganze wäre so einfach. Ohne deine Lösung hätte ich den Speicher eh schon entsorgt.

D.h. ich müsste über die offizielle MQTT - Schnittstelle gehen, wenn ich den Flashspeicher nicht auf Dauer zerstören möchte?

tomquist, 2025-05-17:
> Warum nicht einfach eine lokale MQTT Anbindung direkt onBoard und das ganze wäre so einfach.

Der Speicher hat ja eine lokale MQTT Schnittstelle. Die war auch mal offiziell. Hiermit kann man die relativ einfach nutzen: https://github.com/tomquist/hm2mqtt?tab=readme-ov-file#prerequisites

RALFISBORN, 2025-05-17:
@tomquist cool, das ist ja interessant und wusste ich gar nicht. Und damit habe ich dann nicht mehr das Problem mit dem Flash Speicher beschreiben?

Habe gerade versucht es einzurichten => Handykopplung hat geklappt und angeblich hat er auch die MQTT Settings übernommen. Als Typ wird HMA-3 im Tool angezeigt. Muss ich dann das im Add-on eintragen oder HMA-2 wie im der doku beschrieben? Hab nen Greensolar V2. Allerdings baut er noch keine Verbindung auf. Könnte es daran liegen, dass er aktuell zu leer ist (5%)?

tomquist, 2025-05-17:
Ich gehe mal davon aus mit "Tool" meinst du [das hier](https://tomquist.github.io/hame-relay/b2500.html). Immer das eingeben was das Tool anzeigt.

RALFISBORN, 2025-05-17:
@tomquist ja genau die meinte ich. Ich hab mittlerweile auch die Verbindung hinbekommen (habs nochmal per Chrome getestet, dann hats geklappt). Zumindest habe ich paar Werte wie die Zellspannungen z.B..
Aber paar Werte werden noch nicht angezeigt:

[image]

An was könnte das liegen? Oder muss ich da einfach mal warten, bis der Speicher wieder mehr geladen ist?

Und noch zwei Fragen zur Ansteuerung:
- mit dieser Lösung wird quasi der Flashspeicher nicht mehr ständig beschrieben und ich kann bedenkenlos alle 5s z.B. den Output Wert verändern?
- bei der ESP32 Variante gab es andere Befehl bzgl. des Lademodus (z.B. LoadFirst). Das sind jetzt diese hier oder:

[image]