tomquist/esphome-b2500 #96: Marstek B2500 ESP32C

AHahne64, 2025-04-02:
The firmware file is not created

All values are entered. But an error occurs during creation.

**Expected behavior**

**Actual behavior**

**Additional context**

**Screenshots**

**Device information**

- ESP Board: [e.g. ESP32-C3]

- B2500 Brand: [e.g. Marstek]

- B2500 Version: [e.g. v1, v2, v3]

- Build: [jumpy-angry-kangaroo-173]

**Configuration**

```json
{
  "template_version": "v2",
  "name": "b2500",
  "friendly_name": "B2500",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "enabled": true,
    "topic": "b2500",
    "broker": "mqttbroker.local",
    "port": 1883,
    "username": "",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "Hahne-Smarthome",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "esp32c3",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "enable_reset_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "native_api": {
    "enabled": true
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "B2500",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2025-04-02:
This is the error:
> Option 'variant' does not match selected board.

The reason most likely is that you entered a wrong board identifier. Try setting board to `esp32-c3-devkitm-1` instead.

AHahne64, 2025-04-03:
Hello,

I tested the whole thing again last night. I tried the change you made, but unfortunately that didn't work either.

What finally helped was the change from “Native ESPHome component” to “Minimal native ESPHome component”. 
After this setting it was possible to create the image.

Unfortunately, an error message now appears when flashing that there is not enough memory available.
The settings for memory and ESP were correct with 4 MB.

I then tried the whole thing with an ESP32 where I also get the message that there is not enough memory.
I have now ordered the ESP that you linked to on your page. It will arrive tomorrow and I will try it with this one.

AHahne64, 2025-04-06:
Hello,

now it is up and running.

Thank you for your support.

closed 2025-04-06 by AHahne64