tomquist/esphome-b2500 #98: Missing Topics from Timer Settings

Dropsmasta, 2025-04-07:
Hi tomquist,
mir ist aufgefallen, das die Topics für die Start und End Time nicht vorhanden sind.

Mein Speicher:
[image]
[image]

In der Config.YAML sind sie alle aufgeführt (siehe Code ganz unten), aber in HomeAssistant sind nur die switch und number Entitäten der Timer Settings zu finden:

[image]

Mich hatte es gewundert, dass mein ESP32-WROOM32 ständig Warnings ausgibt - hatte aber eigentlich vermutet, dass es am unzureichenden RAM liegt und mir deswegen bereits ein ESP32-S3 bestellt.

> [22:07:15][W][component:170]: Component mqtt cleared Warning flag
> [22:07:29][W][b2500:305]: Unknown command
> [22:07:29][W][b2500:235]: Failed to receive packet
> [22:07:30][W][b2500:305]: Unknown command
> [22:07:30][W][b2500:235]: Failed to receive packet
> [22:07:30][W][component:237]: Component esp32_ble took a long time for an operation (58 ms).
> [22:07:30][W][component:238]: Components should block for at most 30 ms.

Dann bin ich darauf gestoßen das die oben genannten Topics nicht vorhanden sind, vielleicht ist das Grund warum hier ständig
> Unknown command
> Failed to receive packet

in den Logs ausgegeben wird?

Grüße!

```
datetime:
  - platform: b2500
    b2500_id: b2500_1
    generation: 2
    timer1_start:
      id: b2500_timer1_time_start_1
      name: "B1: Timer 1 Start Time"
      state_topic: "b2500/1/timer/1/start"
      command_topic: "b2500/1/timer/1/start/set"
    timer1_end:
      id: b2500_timer1_time_end_1
      name: "B1: Timer 1 End Time"
      state_topic: "b2500/1/timer/1/end"
      command_topic: "b2500/1/timer/1/end/set"
    timer2_start:
      id: b2500_timer2_time_start_1
      name: "B1: Timer 2 Start Time"
      state_topic: "b2500/1/timer/2/start"
      command_topic: "b2500/1/timer/2/start/set"
    timer2_end:
      id: b2500_timer2_time_end_1
      name: "B1: Timer 2 End Time"
      state_topic: "b2500/1/timer/2/end"
      command_topic: "b2500/1/timer/2/end/set"
    timer3_start:
      id: b2500_timer3_time_start_1
      name: "B1: Timer 3 Start Time"
      state_topic: "b2500/1/timer/3/start"
      command_topic: "b2500/1/timer/3/start/set"
    timer3_end:
      id: b2500_timer3_time_end_1
      name: "B1: Timer 3 End Time"
      state_topic: "b2500/1/timer/3/end"
      command_topic: "b2500/1/timer/3/end/set"
    timer4_start:
      id: b2500_timer4_time_start_1
      name: "B1: Timer 4 Start Time"
      state_topic: "b2500/1/timer/4/start"
      command_topic: "b2500/1/timer/4/start/set"
    timer4_end:
      id: b2500_timer4_time_end_1
      name: "B1: Timer 4 End Time"
      state_topic: "b2500/1/timer/4/end"
      command_topic: "b2500/1/timer/4/end/set"
    timer5_start:
      id: b2500_timer5_time_start_1
      name: "B1: Timer 5 Start Time"
      state_topic: "b2500/1/timer/5/start"
      command_topic: "b2500/1/timer/5/start/set"
    timer5_end:
      id: b2500_timer5_time_end_1
      name: "B1: Timer 5 End Time"
      state_topic: "b2500/1/timer/5/end"
      command_topic: "b2500/1/timer/5/end/set"
```

baal86, 2025-04-12:
I have observed something similar after the upgrade from 220 to 224. They seem to have changed some things in the BLE protocol. I also get 

`[22:07:29][W][b2500:305]: Unknown command
[22:07:29][W][b2500:235]: Failed to receive packet`

messages but previously (v220.12) common error messages regarding WiFi SSID not being read correctly are gone now. I run on the S3 and with quite a minimal configuration of esphome-b2500. In my operating scenario it doesn't really affect me. Is there any documentation on how the protocol was originally reverse-engineered? If I have the tools for it I might sit down and have a closer look at the new features. v224 looks much more stable when using WiFi and BLE so I think it makes sense to look at that version a bit more in detail.

tomquist, 2025-04-12:
@Dropsmasta HomeAssistant unterstützt wohl datetime fields nicht über MQTT. Du kannst mal versuchen den ESP über die Native API statt MQTT einzubinden. Im Config Generator MQTT abschalten und dann Native API einschalten, also ungefähr so:

[image]

Dropsmasta, 2025-05-16:
Vergessen zu antworten:

Habe den ESP32 gewechselt und bin in dem Zuge auch auf die Native API umgestiegen. Funktioniert prima. 👌🏻

Dropsmasta, 2025-05-16:
> I have observed something similar after the upgrade from 220 to 224. They seem to have changed some things in the BLE protocol. I also get
> 
> `[22:07:29][W][b2500:305]: Unknown command [22:07:29][W][b2500:235]: Failed to receive packet`
> 
> messages but previously (v220.12) common error messages regarding WiFi SSID not being read correctly are gone now. I run on the S3 and with quite a minimal configuration of esphome-b2500. In my operating scenario it doesn't really affect me. Is there any documentation on how the protocol was originally reverse-engineered? If I have the tools for it I might sit down and have a closer look at the new features. v224 looks much more stable when using WiFi and BLE so I think it makes sense to look at that version a bit more in detail.
> 
> **Edit:** Using the native integration, not MQTT. So I can confirm the messages are present with V224 and a B2500 build from 2025-04-11 even without any MQTT shenanigans.

I still have similar log entrys, even while using the native API.

closed 2025-05-16 by Dropsmasta