tomquist/esphome-b2500 #100: maybe Voltage value posible ?

sibelle-labs, 2025-04-08:
Hi Tom,

first of all, thank you for your awesome project – it’s working great with my B2500!

I was wondering if it would be possible to expose the battery voltage (e.g. voltage) as a native sensor in a future version of the b2500 component – if the B2500 provides this value via BLE at all.

I'm using the battery together with OpenDTU-OnBattery, and it would be helpful to have the voltage available for additional control or diagnostics.

Thanks again and best regards!

sibelle-labs, 2025-04-11:
Hi Tom,
thanks again for your response!

As mentioned here:
👉 https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1835#discussioncomment-12766798
the individual cell voltages (e.g. 16 × 3.3 V = 52.8 V) are simply summed up to calculate the total voltage.

Since you're already reading each cell's value, I guess this part is already handled — or am I missing something? 😊

thanks a lot !

Best regards!

tomquist, 2025-04-11:
Ah right, almost forgot about this. The topic b2500/{storage}/battery/cell_voltage even exposes this value through the parameter `sum`, so no need to create a helper. It’s not exposed as a Home Assistant sensor right now but it should be easy to add this sensor manually.

sibelle-labs, 2025-04-11:
you are the best !
thanks a lot :-)

closed 2025-04-12 by sibelle-labs