tomquist/esphome-b2500 #105: Build ESPHome Firmware #3593 fails [quick-wobbly-dolphin-93]

MichelLMAA, 2025-04-12:
**Describe the issue**

The build process failed.
With only one Device an V1 it was build correctly.

**Expected behavior**

**Actual behavior**

**Additional context**

**Screenshots**

**Device information**

- ESP Board: esp32-s3-devkitc-1

- B2500 Brand: Marstek 

- B2500 Version: V2, is the Version related to the FW Version or HW Version?  
- Build: [quick-wobbly-dolphin-93]

**Configuration**

```json
{
  "template_version": "v2",
  "name": "b2500-mhr",
  "friendly_name": "B2500-MHr",
  "poll_interval_seconds": 5,
  "log_level": "DEBUG",
  "mqtt": {
    "enabled": true,
    "topic": "b2500",
    "broker": "192.168.178.59",
    "port": 1883,
    "username": "",
    "password": "***",
    "discovery": false
  },
  "wifi": {
    "ssid": "WLAN_Motte",
    "password": "***"
  },
  "board": "esp32-s3-devkitc-1",
  "variant": "esp32s3",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "enable_reset_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "native_api": {
    "enabled": false
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "B2500",
      "version": 2,
      "mac_address": "***",
      "id": "***"
    },
    {
      "name": "B2500-2",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2025-04-12:
Das ist der Build Fehler:
> Error: The program size ([1857217](tel:1857217) bytes) is greater than maximum allowed ([1835008](tel:1835008) bytes)

Welchen ESP32 hast du genau und wie viel Flash Speicher hat der? Wenn er 8MB Flash hat, kannst du unter Advanced einfach die Flash Size entsprechend anpassen. Ansonsten habe ich hier ein paar Tipps geschrieben um die Image Größe zu reduzieren: https://github.com/tomquist/esphome-b2500/discussions/104#discussioncomment-12795427

MichelLMAA, 2025-04-12:
Hi Tomquist, 
danke für dei schnelle Antwort! es ist ein ESP32-S3-WROOM-1-N16R8  also 16MB Flash, ich werde mal die Flashsize anpassen und einen neuen build anwerfen .

Ein Frage noch, in der Eigabemaske für den Batteriespeicher wird die Version (1 oder 2) abgefragt bezieht sich das auf die typen HMA HMB?  oder Firmware Version? 

Übringens, sehr cooles Projekt!

tomquist, 2025-04-12:
V1 = Version ohne Timer und regelbarer Ausgangsleistung
V2 = Version mit Timer und regelbarer Ausgangsleistung

Es wird heute nur noch V2 verkauft. Wenn Du in der App Timer konfigurieren kannst, hast Du eine V2.

closed 2025-10-20 by tomquist