tomquist/esphome-b2500 #111: Marstek HMJ  Firmware 106

Dirk-Ahoy, 2025-04-16:
Moin zusammen, ich hab per Support die FW 106.20 bekommen für den HMJ

Update hat auch nach einigen Versuchen geklappt.  

Leider gibts jetzt ein paar Fehler im log

[code]
09:51:14
[I]
[esp-idf:000][1;31m[BTU_TASK][0;32m
[0;31mE (103751) BT_BTM: BTM_BleScan scan not ac
09:51:14
[I]
[esp-idf:000][1;31m[BTU_TASK][0;32m
[0;33mW (103754) BT_APPL: bta_dm_ble_scan stop scan failed, status
09:51:14
[I]
[esp32_ble_client:110]
[0] [98:A1:4A:3E:3D:B8] 0x00 Attempting BLE connection
09:51:15
[I]
[b2500:015]
Connected successfully!
09:51:17
[I]
[esp32_ble_client:337]
[0] [98:A1:4A:3E:3D:B8] Connected
09:51:17
[I]
[main:023]
Connected to BLE device 1 (B2500)
09:51:22
[W]
[b2500:305]
Unknown command
09:51:22
[W]
[b2500:235]
Failed to receive packet
09:51:25
[W]
[b2500:173]
SSID is empty or invalid
09:51:25
[W]
[b2500:282]
Failed to parse wifi info
09:51:25
[W]
[b2500:235]
Failed to receive packet
09:51:26
[W]
[b2500:305]
Unknown command
09:51:26
[W]
[b2500:235]
Failed to receive packet
[/code]

Kann ich irgendwie helfen das man das gefixt bekommt?  Vermutlich liegt es an den neuen Funktionen.

hydrabanjo, 2025-04-17:
Moin! Bei mir ist das gleiche Problem, ebenfalls mit einem HMJ, SW Version 106:
`09:54:58 [W] [b2500:173] SSID is empty or invalid
09:54:58 [W] [b2500:282] Failed to parse wifi info
09:54:58 [W] [b2500:235] Failed to receive packet
09:54:59 [W] [component:239] Component esp32_ble took a long time for an operation (54 ms).
09:54:59 [W] [component:240] Components should block for at most 30 ms.
09:55:08 [W] [b2500:173] SSID is empty or invalid
09:55:08 [W] [b2500:282] Failed to parse wifi info
09:55:08 [W] [b2500:235] Failed to receive packet`

Aktuell werden mir alle benötigten Daten in MQTT angezeigt. Die Fehler wiederholen sich dann immer wieder im Log.

hydrabanjo, 2025-04-17:
Bei mir ist es auch so, dass der ESP sporadisch auf einmal keine Daten mehr liefert. Der Bluetooth und auch der WLAN Empfang sollten gut genug sein.

Marc9485, 2025-04-24:
Hey, bei #115 haben wir das gleiche Thema

RicardoHipp, 2025-04-26:
Bei mir auch das gleich e Thema:

10:16:20
[W]
[b2500:282]
Failed to parse wifi info
10:16:20
[W]
[b2500:235]
Failed to receive packet
10:16:25
[W]
[b2500:173]
SSID is empty or invalid

tomquist, 2025-04-26:
Könnte irgendjemand Logs hier mal unter Advanced den Log Level "DEBUG" stellen und danach logs hier posten? Das würde helfen das Problem zu beheben.

tomquist, 2025-04-26:
Habe mittlerweile die Logs. Sieht so aus als würde der Speicher in der neuen Version die SSID nicht mehr rausgeben. Die Meldungen sind nur eine Warnung und können vorerst ignoriert werden.

sschw56, 2025-08-29:
Hi,

ich habe die V2 des Speichers mit der FW 110 und das gleiche Problem. 
Das unschöne ist, dass mein HA Log voll ist mit der Meldung und ich auch an anderer Stelle keine Warnung des ESPs ausgeben kann, da alles zugespamt wird. 
Es hilft leider auch nicht die zwei wifi-bezogenen Sensoren aus der yaml zu nehmen, da die Meldung wohl aus dem Code selbst kommen. Könnte man ggf die Meldung entweder auf eine Info herausrufen oder per Schalter einstellbar machen?

Viele Grüße und Danke für all die tolle Arbeit!!