tomquist/esphome-b2500 #120: Restarting ESP32 Sets values on B2500

strangesource, 2025-04-27:
I have observed that the ESP32 sometimes restarts for no obvious reason - which would not be a huge problem as it seems to recover nicely. It seems to change some of the values on the B2500, notably the "Selbstanpassung" to "Manual".

Is this a known issue?

Not sure if that efforts anything, I have three batteries connected and the firmware is v224.

vinzent142, 2025-05-18:
Changed my setup to "Selbstanpassung" yesterday and observed the same as @strangesource. Upon booting the ESP32-S3 it switches the battery to "Manual".

I use just one B2500 with version v226.

pr3mi3r3, 2025-05-21:
Same problem with HMJ-2.
Upon booting the ESP32-S3 it switches the battery to "Manual".

tomquist, 2025-06-22:
Duplicate of #122