tomquist/esphome-b2500 #122: Smart Meter Enabled Switch turns itself off

konterr, 2025-04-29:
Hi,
I have two Marstek b2500 which I am trying to integrate to homeassitant. The connection with the ESP works fine and I get data. As soon as I want to activate the smart meter connection though, it turns it off after a few minutes (using your smart meter addon - it is great, thank you!). It seems data is not send correctly in that timeframe either. Might the ESP32-wroom be the problem (you just answered my build problem) as it is quite a bit of data send?
Here two screenshots of the weird data:
[image]

[image]

When the line in the one graph goes up, the smart meter connection was turned off. Output watts showed a correct value as well in the scheduled mode. I unplugged the ESP for now so the smart meter connection works.

The marstek is running firmware V108.

Kind regards,
Konterr

konterr, 2025-04-29:
Oh, the Logs seemed uneventful. I had some of those SSID warnings, and connection errors caused by unplugging.

Niboo90, 2025-04-30:
Same problem here, as soon as I connect the ESP to the grid, the discharge setting in the Martsek is changing from smart meter to manual -.-

oktabruna, 2025-05-01:
The B2500 model is the device type HMJ-2

In my case, it switches between simultaneous charge , discharge and load first.

Niboo90, 2025-05-05:
does anyone have an solution for the Probelm?

Niboo90, 2025-05-10:
Hast du auch 2 Speicher konfiguriert?

goldrapper19872504, 2025-05-10:
Ja, einmal einen HMJ und einen HMA. Aber ohne Webserver kamen die Daten unregelmäßig bis gar nicht. Habe jetzt ein Post gesehen von TomQuist wo er schreibt das man bei dem ESP 32 S3 8MB Flash auswählen soll und mqtt ausschalten bei Home Asisstant dafür aber API anschalten und seitdem arbeitet er störungsfrei.

Niboo90, 2025-05-11:
Gerade nochmal getestet, sobald der ESP sich mit dem Speicher verbindet resettet er mir die SmartMeter Einstellung auf manual. APi ist aktiv und WebServer ist aus. Hab 2 B2500 mit dem ESP verbunden.

konterr, 2025-05-12:
Ich habe einen HMA-1 und einen HMJ-2. Beim HMA-1 habe ich das Problem, dass sich der Speicher immer auf den Manual Modus zum entladen stellt. Ich kann die DoD bspw. problemlos anpassen... Es ist auch egal ob ich das über die Marstek App einschalte, oder über Homeassistant. Nach ein paar Sekunden stellt sich das wieder um. Der Speicher Arbeitet aber z.T. kurzzeitig korrekt in dem Modus.
Ich will die Speicher vor allem gar nicht darüber steuern, das machen sie über den Adaptive mode ganz gut alleine. Ich will nur die Daten haben :D

Als ich habe mir mal einen der empfohlenen ESPs zugelegt. Webserver und MQTT ist deaktiviert. Die Verbindung läuft auch problemlos, soweit ich das beurteilen kann.

konterr, 2025-05-13:
@Niboo90 Ich habe in der Config die Speicher einmal getauscht. Jetzt kann ich bei einem Speicher das DoD nicht mehr anpassen, aber das Smart Meter läuft bei beiden Speichern :D Da kann ich mit leben, auch wenn ich das nicht ganz verstehe.

Dropsmasta, 2025-05-16:
> The B2500 model is the device type HMJ-2
> 
> In my case, it switches between simultaneous charge , discharge and load first.

War bei mir genau so. Der HMJ ist spontan auf "Loadfirst" gewechselt.

Kam zum Glück aber nicht regelmäßig vor

Niboo90, 2025-05-16:
@konterr ich hatte durch Zufall mal die Timer beim Betroffenen Speicher deaktiviert, seitdem stellt er nicht mehr den Entladungsmodus um.

pr3mi3r3, 2025-05-19:
> [@Niboo90](https://github.com/Niboo90) Ich habe in der Config die Speicher einmal getauscht. Jetzt kann ich bei einem Speicher das DoD nicht mehr anpassen, aber das Smart Meter läuft bei beiden Speichern :D Da kann ich mit leben, auch wenn ich das nicht ganz verstehe.

Hat bei mir nicht funktioniert leider...
Smart Meter sofort auf off...

Wie ist dein komplettes Setup? 
Welches Board nutzt du?
Welche Config für den ESP?

TheFarmer88, 2025-05-20:
Ich habe auch einen HMJ-2 und hatte das gleiche Problem. 

Beobachte mal über den ESPHome Builder im Homeassistant das Log des ESP.
Bei mir hat der ESP ständig versucht die Zeiten für Timer 4 + 5 neu zu setzen und dadurch die Selbstanpassung deaktiviert. Ich habe daraufhin die Zeiten für alle 5 Timer im Marstek hinterlegt und das Problem war weg.

[image]

tomquist, 2025-06-22:
I just pushed a potential fix. I'm currently not able to test this so I'd appreciate if anyone could give the latest version a try.