tomquist/esphome-b2500 #123: failed build 03052025 21:50

Hector-SaTe, 2025-05-03:
**The build fails. I tried 2 times:**

[B2500 [jumpy-jumpy-toucan-960]](https://github.com/tomquist/esphome-b2500/actions/runs/14814025341)

[B2500 [quirky-clumsy-flamingo-375]](https://github.com/tomquist/esphome-b2500/actions/runs/14813914820)

**Device information**

- ESP Board: [e.g. ESP32-C3]

- B2500 Brand: [e.g. Marstek]

- B2500 Version: [e.g. v1, v2, v3]

**Configuration**

```json
{
  "template_version": "v2",
  "name": "b2500-x1",
  "friendly_name": "B2500-X1",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "enabled": true,
    "topic": "b2500",
    "broker": "mqttbroker.local",
    "port": 1883,
    "username": "mqtt_user",
    "password": "***",
    "discovery": true
  },
  "wifi": {
    "ssid": "LaSNiVHeS",
    "password": "***"
  },
  "board": "esp32dev",
  "variant": "auto",
  "flash_size": "4MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "enable_reset_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": true,
  "powerzero": {
    "grid_power_topic": "sensor.shellypro3em_total_active_power",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "native_api": {
    "enabled": true
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "B2500-X1",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2025-05-03:
Exactly the same issue as here: https://github.com/tomquist/esphome-b2500/issues/121#issuecomment-2839920781

Hector-SaTe, 2025-05-04:
Thanks, it worked now. I will try using the ESP32 now.

BTW I would like to try the BLE connection directly from my OrangePi5. There I have my HA running in docker. That way I wouldn't need to use the ESP32. I am trying to get notifications from the B2500, using commands and UUID from your repo and the help of COPILOT to translate it to python...  here some example:
https://github.com/Hector-SaTe/hm2500/blob/master/b2500_ble_tomquist_v1.py
The other files are also trying to connect using gadgetbridge logic.

I do not get any response from it. I do not know BLE at all so I am helpless here XD.