tomquist/esphome-b2500 #124: Probleme beim Flashen

daewoo42, 2025-05-05:
Build bauen funktioniert, aber mein ESP32 läuft immer in eine Boot-Schleife.

`[09:14:28]ESP-ROM:esp32s3-20210327
[09:14:28]Build:Mar 27 2021
[09:14:28]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[09:14:28]Saved PC:0x40043ac8
[09:14:28]SPIWP:0xee
[09:14:28]mode:DIO, clock div:1
[09:14:28]load:0x3c140020,len:0x6a7dc
[09:14:28]load:0x3fc9a500,len:0x4378
[09:14:28]load:0x40374000,len:0x1494
[09:14:28]load:0x42000020,len:0x1360a0
**[09:14:28]Invalid image block, can't boot.**
[09:14:28]ets_main.c 329 
[09:14:28]ESP-ROM:esp32s3-20210327
[09:14:28]Build:Mar 27 2021
[09:14:28]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[09:14:28]Saved PC:0x40043ac8
[09:14:28]SPIWP:0xee
[09:14:28]mode:DIO, clock div:1
[09:14:28]load:0x3c140020,len:0x6a7dc
[09:14:28]ESP-ROM:esp32s3-20210327
`
Hier meine Config:

esphome:
  name: "b2500"
  friendly_name: "B2500"
  on_boot:
    priority: -100
    then:
      - lambda: id(b2500_device_generation_1).publish_state("2");
      - lambda: id(b2500_device_name_1).publish_state("B2500");

esp32:
  board: "esp32-s3-devkitc-1"
  variant: "esp32s3"
  flash_size: 4MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      COMPILER_OPTIMIZATION_SIZE: y

# Enable logging
logger:
  level: INFO
ota:
  - platform: esphome
    password: "my_ota_password"
external_components:
  - source:
      type: git
      url: https://github.com/tomquist/esphome-b2500
      ref: "main"
    refresh: 0s
    components: [b2500]

wifi:
  ssid: "xxxxxxxx"
  password: "xxxxxxxxxxxxxxxxxxxxx"
  reboot_timeout: 0s
  fast_connect: True
  manual_ip:
    static_ip: "192.168.178.40"
    gateway: "192.168.178.1"
    subnet: "255.255.255.0"
    dns1: "192.168.178.1"
  on_connect:
    - logger.log:
[... 486 lines cut ...]
      accuracy_decimals: 0
  - platform: total_daily_energy
    name: "B2500 - 1 - B2500: Total Daily Energy In"
    id: b2500_total_daily_energy_in_1
    state_topic: "b2500/1/pv/energy"
    power_id: b2500_total_power_in_1
    unit_of_measurement: kWh
    method: left
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      - multiply: 0.001
  - platform: total_daily_energy
    name: "B2500 - 1 - B2500: Total Daily Energy Out"
    id: b2500_total_daily_energy_out_1
    state_topic: "b2500/1/power/energy"
    power_id: b2500_total_power_out_1
    unit_of_measurement: kWh
    method: left
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      - multiply: 0.001

daewoo42, 2025-05-05:
Ich habe den ESP32S3 aus dem Link verwendet und auch schon viele verschiedene Konfig's getestet.
Ich bin mir ziemlich sicher, dass der Fehler vermutlich vor dem PC sitzt, aber ich komme nicht drauf, was ich falsch mach.
Ich habe sowohl beim Flashen über die COM wie auch über USB das gleiche Ergebnis.

tomquist, 2025-05-06:
Hmm, schwer zu sagen was da schief läuft. Der Crash scheint in den Logs zu fehlen. Versuche vielleicht mal unter Advanced „Enable DIO Flash Mode“ zu aktivieren. Außerdem ist der Flash auf 4MB gestellt. Eventuell mal erhöhen auf den entsprechenden Wert deines Boards.

daewoo42, 2025-05-07:
Ich sag ja, das Problem sitzt vor dem PC.
Der von der verlinkte ESP32S3 hat natürlich **8MB**.
Ich werde es am WE testen, sobald ich Zeit hab.
Feedback folgt.

closed 2025-06-30 by tomquist