tomquist/esphome-b2500 #127: Werte beim HMJ-2 (V108) werden nicht aktzeptiert  - Automation für HMJ-2 vs. HMA-1

Dropsmasta, 2025-05-13:
Hi tomquist,

habe gestern einen weiteren Marstek B2500 erhalten und habe die Gelegenheit genutzt um auf ein ESP32-S3 N16R8 zu wechseln (das alte ESP32 ist aus bekannten Gründen ständig abgestürzt...).

Nun habe ich hier zwei unterschiedliche Hardware-Versionen mit zwei unterschiedlichene Firmeware-Ständen zu stehen:
Einmal einen HMJ-2 ("B1") mit V108 und  einen HMA-1 ("B2") mit V224.

_(Vorab: Die Automation ist nur eine Behelfslösung, weil meine beiden B2500 weder meinen Shelly Pro 3EM erkennen wollen, noch die Emulation über dein B2500-Meter Add-On - aber das ist ein anderes Thema was hier nicht diskutiert werden soll. )_

Ich habe eine Automation zu laufen die den Timer 5 aktiviert und die Output Power des Timer 5 auf 600W stellt.
Das ganze funktioniert mit dem dem HMA-1 super, aber beim HMJ-2 werden die gesetzten Werde nicht akzeptiert/wieder zurück gesetzt. (Siehe Screenshot "Verhalten").
Gleiches Verhalten ist zu sehen wenn ich manuell den Timer aktiviere oder den Output des Timers verändere.

Habe die Info-Logs aus dem Zeitraum mit angehangen.

Falls ich Debug-Logs nachliefern soll, gib gerne Bescheid.

Ist das schon bekannt?

Verhalten:

[image]

Daten:

[image]

Info-Logs vom ESP32:

```
[12:09:42][W][b2500:173]: SSID is empty or invalid
[12:09:42][W][b2500:282]: Failed to parse wifi info
[12:09:42][W][b2500:235]: Failed to receive packet
[12:09:46][W][b2500:305]: Unknown command
[12:09:46][W][b2500:235]: Failed to receive packet
[12:10:00][W][b2500:173]: SSID is empty or invalid
[12:10:00][W][b2500:282]: Failed to parse wifi info
[12:10:00][W][b2500:235]: Failed to receive packet
[12:10:01][W][main:053]: Free heap: 137320
[12:10:02][W][b2500:173]: SSID is empty or invalid
[12:10:02][W][b2500:282]: Failed to parse wifi info
[12:10:02][W][b2500:235]: Failed to receive packet
[12:10:12][W][b2500:305]: Unknown command
[12:10:12][W][b2500:235]: Failed to receive packet
[12:10:17][W][b2500:305]: Unknown command
[12:10:17][W][b2500:235]: Failed to receive packet
[12:10:19][W][b2500:173]: SSID is empty or invalid
[12:10:19][W][b2500:282]: Failed to parse wifi info
[12:10:19][W][b2500:235]: Failed to receive packet
[12:10:22][W][b2500:173]: SSID is empty or invalid
[12:10:22][W][b2500:282]: Failed to parse wifi info
[12:10:22][W][b2500:235]: Failed to receive packet
[12:10:40][W][b2500:173]: SSID is empty or invalid
[12:10:40][W][b2500:282]: Failed to parse wifi info
[12:10:40][W][b2500:235]: Failed to receive packet
[12:10:42][W][b2500:173]: SSID is empty or invalid
[12:10:42][W][b2500:282]: Failed to parse wifi info
[12:10:42][W][b2500:235]: Failed to receive packet
[12:10:59][W][b2500:173]: SSID is empty or invalid
[12:10:59][W][b2500:282]: Failed to parse wifi info
[12:10:59][W][b2500:235]: Failed to receive packet
[12:11:01][W][b2500:305]: Unknown command
[12:11:01][W][b2500:235]: Failed to receive packet
[12:11:01][W][main:053]: Free heap: 137320
[12:11:02][W][b2500:173]: SSID is empty or invalid
[12:11:02][W][b2500:282]: Failed to parse wifi info
[12:11:02][W][b2500:235]: Failed to receive packet
[12:11:06][W][b2500:305]: Unknown command
[12:11:06][W][b2500:235]: Failed to receive packet
[12:11:20][W][b2500:173]: SSID is empty or invalid
```

Dropsmasta, 2025-05-14:
Nachtrag:

Habe meine Automation auf den Timer 1 gewechselt, jetzt funktioniert es mit beiden Akkus (siehe Screenshot).

_(Der HMA-1 empfängt auch die Daten vom B2500-Meter Add-On, der HMJ-2 aber noch nicht)_

[image]

Dirk-Ahoy, 2025-05-16:
Firmware 1.08 hat auch einen neuen Parameter in der App " Überschusseinspeisung"   Das soll wohl ein Bypass sein wenn der Akku fast voll ist das PV komplett durchgereicht wird.   Scheint auch zu klappen aber in der Software ist ja noch kein Feld dafür drin.

Dropsmasta, 2025-05-16:
Interessant! Habe ich noch garnicht gesehen.

Ich habe jetzt zusätzlich das Verhalten, das die Zeitplanung 2 Stunden zu früh anspringt, obwohl beide Akkus die korrekte Zeit ausgeben. 🙈

Der neue HMJ hatte bis gestern eine Stunde Zeitunterschied zum alten HMA (beide auf Germany-Berlin gestellt) und trotzdem hat er genau wie der andere zu früh eingespeist (entsprechend eine Stunde zu früh zu seiner Systemzeit).

Dirk-Ahoy, 2025-05-16:
Genau das hab ich auch schon bemerkt heut morgen hat er aufeinmal aufgehört obwohl der Timer noch lief - da muss ich noch mal genauer schauen.

citpanys, 2025-06-30:
Ich habe das gleiche Problem mit meinem HMJ-2. Wenn ich per Home Assistant (manuell oder per Automation) das oder Webserver die Entladeleistung setze, wird sie ca. 5 Sek. später wieder auf den vorherigen Wert zurück gesetzt. Vermutlich auf den Wert, den ich zuletzt in der Marstek App gesetzt habe. 

Habs auch schon mal mit der V2 statt V1 versucht (woran erkennt man eigentlich, ob V1 oder V2 nötig ist?) - ohne Erfolg. 

Ich weiß echt nicht mehr weiter.

maxytmaxyt, 2026-09-25:
@claude