tomquist/esphome-b2500 #128: Marstek2Akkusystem

Swamartob, 2025-05-14:
**Describe the issue**

**Steps to reproduce**

**Expected behavior**

**Actual behavior**

**Additional context**

**Screenshots**

**Device information**

- ESP Board: [e.g. ESP32-C3]

- B2500 Brand: [e.g. Marstek]

- B2500 Version: [e.g. v1, v2, v3]

- Build: [happy-sleepy-tiger-420]

**Configuration**

```json
{
  "template_version": "v2",
  "name": "b2500akku1",
  "friendly_name": "B2500",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "enabled": true,
    "topic": "Akku",
    "broker": "mqttbroker.local",
    "port": 1883,
    "username": "mqtt-user",
    "password": "***",
    "discovery": true
  },
  "wifi": {
    "ssid": "IOTHome",
    "password": "***"
  },
  "board": "esp32-s3-devkitc-1",
  "variant": "esp32s3",
  "flash_size": "8MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": true,
  "set_wifi": {
    "ssid": "IOTHome",
    "password": "***"
  },
  "enable_set_mqtt": true,
  "enable_reset_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": true,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "native_api": {
    "enabled": true
  },
  "enable_dio_flash_mode": true,
  "storages": [
    {
      "name": "Akku1/0525",
      "version": 2,
      "mac_address": "***"
    },
    {
      "name": "Akku2/0525",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2025-05-14:
What is the issue? The build seems to have succeeded.

Swamartob, 2025-05-14:
Ist auch schon mein 5 Versuch oder so! 
Ich versuche ihre Version morgen zu Flashen. leider weiß ich nicht wie der Download der *.bin Datei funktioniert.
Morgen wird der von ihnen empfohlene ESP32 geliefert.