tomquist/esphome-b2500 #132: Compiling not posible

urgti, 2025-05-24:
All compile tries will fail after 1 minute.

**Steps to reproduce**

**Expected behavior**

**Actual behavior**

**Additional context**

**Screenshots**

**Device information**

- ESP Board: [e.g. ESP32-C3]

- B2500 Brand: [e.g. Marstek]

- B2500 Version: [e.g. v1, v2, v3]

- Build: [fluffy-silly-platypus-681]

**Configuration**

```json
{
  "template_version": "v2",
  "name": "b2500",
  "friendly_name": "B2500",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "enabled": true,
    "topic": "b2500",
    "broker": "mqttbroker.local",
    "port": 1883,
    "username": "mqtt-user",
    "password": "***",
    "discovery": true
  },
  "wifi": {
    "ssid": "iot",
    "password": "***"
  },
  "board": "esp32-s3-devkitc-1",
  "variant": "esp32s3",
  "flash_size": "8MB",
  "idf_platform_version": "",
  "enable_auto_restart": true,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": true,
  "enable_cmd30": false,
  "enable_esp_temperature": true,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": true,
  "enable_reset_mqtt": true,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.1.100",
    "gateway": "192.168.1.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.1.1"
  },
  "enable_web_server": true,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": true,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": false,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": true
  },
  "native_api": {
    "enabled": true
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "B2500",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

cslphoenix, 2025-05-24:
Abend, ich habe es auch gerade Probiert mit einer config womit ich es damals erstellt habe.
Komisch nur das ich beim Drucken auf Download klicken wollte nach der Erstellung, diese Meldung bekommen habe.

`Once the build completed, download your firmware using this link: [Firmware Download](https://esphome-b2500-images.s3.eu-west-1.amazonaws.com/firmware/bizarre-dark-sloth-908.zip)`

`This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>RC4E9B3TM80QJ4N7</RequestId>
<HostId>GhYIyKhOiUL6A/oe8H5oi12f0kcy8dQ2wayrm+hbhROhoWkL8Bxdy5e1OiNlW6s+8HoWbGsBTfg=</HostId>
</Error>`

Vielleicht hilft es ja :)

tomquist, 2025-05-25:
This is fixed now

closed 2025-05-25 by tomquist