tomquist/esphome-b2500 #150: Invalid image block, can't boot.

LASSENSEDIS, 2025-06-19:
Hallo, 
ich versuche derzeit ein Build zu erstellen mit dem ich meinen B2500-D HMJ-2 (V110.9) in HA Integrieren kann. 
Bin der Anleitung gefolgt, habe den ESP über den Link gekauft und erhalte jedes mal nach dem installieren über ESPHome Web folgende Meldungen im Log:

[21:19:53]ESP-ROM:esp32s3-20210327
[21:19:53]Build:Mar 27 2021
[21:19:53]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[21:19:53]Saved PC:0x40043ac8
[21:19:53]SPIWP:0xee
[21:19:53]mode:DIO, clock div:1
[21:19:53]load:0x3c140020,len:0x55cf0
[21:19:53]load:0x3fc9bf00,len:0x4774
[21:19:53]load:0x40374000,len:0x5b84
[21:19:53]load:0x42000020,len:0x13344c
[21:19:53]Invalid image block, can't boot.
[21:19:53]ets_main.c 329

Meine YAML sieht wie folgt aus: 
esphome:
  name: "b2500"
  friendly_name: "B2500"
  on_boot:
    priority: -100
    then:
      - lambda: id(b2500_device_generation_1).publish_state("1");
      - lambda: id(b2500_device_name_1).publish_state("B2500");

esp32:
  board: "esp32-s3-devkitc-1"
  variant: "esp32s3"
  flash_size: 4MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      COMPILER_OPTIMIZATION_SIZE: y

# Enable logging
logger:
  level: INFO
ota:
  - platform: esphome
    password: "my_ota_password"
external_components:
  - source:
      type: git
      url: https://github.com/tomquist/esphome-b2500
      ref: "main"
    refresh: 0s
    components: [b2500]

wifi:
  ssid: "MEIN ROUTER NAME"
  password: "PASSWORT"
  reboot_timeout: 0s
  fast_connect: True
  on_connect:
    - logger.log:
        level: INFO
        format: "Connected to WiFi. Starting BLE scan."
  on_disconnect:
    - logger.log:
        level: INFO
        format: "Disconnected from WiFi. Stopping BLE scan."
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ESPHome-b2500"
web_server:
  port: 80
  ota: false
[... 341 lines cut ...]
      icon: mdi:thermometer
  - platform: total_daily_energy
    name: "B2500 - 1 - B2500: Total Daily Energy In"
    id: b2500_total_daily_energy_in_1
    state_topic: "b2500/1/pv/energy"
    power_id: b2500_total_power_in_1
    unit_of_measurement: kWh
    method: left
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      - multiply: 0.001
  - platform: total_daily_energy
    name: "B2500 - 1 - B2500: Total Daily Energy Out"
    id: b2500_total_daily_energy_out_1
    state_topic: "b2500/1/power/energy"
    power_id: b2500_total_power_out_1
    unit_of_measurement: kWh
    method: left
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      - multiply: 0.001

LASSENSEDIS, 2025-06-20:
Unfortunately, this setting doesn't change anything. The error remains the same. I also bought a second board through the link. The error occurs with this one as well. Do you have any other ideas?

LASSENSEDIS, 2025-06-20:
1. Go To tomquist.github.io/esphome-b2500/
2. set Board to esp32-s3-devkitc-1 and Variant to esp32s3
3. Fill in the MQTT, Wi-Fi data, and add the MAC address.
4. Press Build Image 
5. Connect ESP via USB C 
6. Connect via Web ESP Home 
7. Then I go to Install and select the OTA file. It completes successfully. Then the same text appears in the log, no matter how many times I try it with a new build.

Use Google Chrome and try it with a MacBook and a Windows PC.

Are i doing something wrong here? According to the label on the ESP, it is an ESP32-S3-N16R8, as I mentioned before, purchased via the link. I think I might be doing something wrong somewhere.

tomquist, 2025-06-20:
Try selecting the other .bin file. The OTA file is for updates over the air only.

synack666, 2025-06-20:
Ich hab den ESP direkt bei HA via USB angeschlossen und ESP Home das Teil einrichten lassen, danach die yaml in esp home -> edit  eingefügt, läuft

LASSENSEDIS, 2025-06-21:
> Using the factory bin is definitely correct. I think the device only appears in ESPHome if you build the image directly through ESPHome. Do you see the device posting into your MQTT broker on the `b2500/` topic? If not, double-check the wifi and mqtt settings.

I did it! I didn't know how to check if it was sending data. I have now manually copied the YAML to the ESP. Then, it had a problem with the configuration, which I was able to fix by disabling the web server, as it apparently had trouble with port 80.

closed 2025-06-22 by LASSENSEDIS