tomquist/esphome-b2500 #165: ca. 20 Images erzeugt mit und ohne error

Thaui57, 2025-07-09:
egal welche config ich kompiliere der log enthält immer den Fehler "[16:38:21]Invalid image block, can't boot."

beim händischen verbinden des Wlans "An error occurred. Improv Wi-Fi Serial not detected"

**Steps to reproduce**

**Expected behavior**

**Actual behavior**

**Additional context**

**Screenshots**

**Device information**

- ESP Board: [e.g. ESP32-S3 N16R8]

- B2500 Brand: [e.g. Marstek]

- B2500 Version: [e.g. v1, v2, v3]

- Build: [bouncy-nutty-rabbit-768]

**Configuration**

```json
{
  "template_version": "v2-minimal",
  "name": "b2500",
  "friendly_name": "B2500",
  "poll_interval_seconds": 5,
  "log_level": "INFO",
  "mqtt": {
    "enabled": true,
    "topic": "marstek",
    "broker": "192.168.0.2",
    "port": 1883,
    "username": "",
    "password": "***",
    "discovery": true
  },
  "wifi": {
    "ssid": "DIR-825",
    "password": "***"
  },
  "board": "freenove_esp32_s3_wroom",
  "variant": "auto",
  "flash_size": "8MB",
  "idf_platform_version": "",
  "enable_auto_restart": false,
  "auto_restart": {
    "restart_after_error_count": 8
  },
  "enable_cellquery": false,
  "enable_timer_query": false,
  "enable_cmd30": false,
  "enable_esp_temperature": false,
  "esp_temperature": {
    "variant": "internal"
  },
  "enable_powermeter": false,
  "enable_experimental_commands": false,
  "enable_hexdump": false,
  "enable_set_wifi": false,
  "set_wifi": {
    "ssid": "MyWifi",
    "password": "***"
  },
  "enable_set_mqtt": false,
  "enable_reset_mqtt": false,
  "powermeter": {
    "tx_pin": "GPIO6",
    "rx_pin": "GPIO7",
    "baud_rate": 9600,
    "stop_bits": 1
  },
  "enable_enforce_dod": false,
  "enable_powerzero": false,
  "powerzero": {
    "grid_power_topic": "tibber-esp/sensor/power/state",
    "limit_cmd_topic": "openDTU/XXXXXXXXXXXX/cmd/limit_persistent_relative",
    "limit_state_topic": "openDTU/XXXXXXXXXXXX/state/limit_relative"
  },
  "enable_manual_ip": false,
  "manual_ip": {
    "ip": "192.168.0.100",
    "gateway": "192.168.0.1",
    "subnet": "255.255.255.0",
    "dns": "192.168.0.1"
  },
  "enable_web_server": false,
  "web_server": {
    "port": 80,
    "js_include": "./v2/www.js"
  },
  "enable_ota": false,
  "ota": {
    "password": "***",
    "enable_unprotected_writes": false
  },
  "enable_fallback_hotspot": false,
  "fallback_hotspot": {
    "ssid": "ESPHome-b2500",
    "enable_captive_portal": false
  },
  "native_api": {
    "enabled": false
  },
  "enable_dio_flash_mode": false,
  "storages": [
    {
      "name": "B2500-1",
      "version": 2,
      "mac_address": "***"
    },
    {
      "name": "B2500-2",
      "version": 2,
      "mac_address": "***"
    }
  ]
}
```

tomquist, 2025-07-09:
Welche Datei has du geflashed?

tomquist, 2025-07-09:
Ah sorry, hab nur ota gelesen. Auf keden Fall kicht das OTA image flashen sondern das factory image.

closed 2025-07-10 by Thaui57