tomquist/esphome-b2500 #168: Keine Kompelierung möglich

JoJoFirst, 2025-07-17:
Seid Vorgestern keine kompelierung mehr möglich.

DerEmder, 2025-07-17:
Ohne Fehlermeldung etwas dürftig, oder?

Anyway, I cannot compile it either since today (one of those ESPHome updates to 2025.7.0):

Failed config:

```
button.b2500: [source /config/esphome/jackie.yaml:161]
  platform: b2500
  b2500_id: b2500_1
  
  not a valid value.
  generation: 2
  reboot: 
    id: b2500_button_reboot_1
    name: B1: Reboot B2500
  factory_reset: 
    id: b2500_button_factory_reset_1
    name: B1: Reset to Factory Settings
    disabled_by_default: True
```

Any idea?

DerEmder, 2025-07-17:
> Seid Vorgestern keine kompelierung mehr möglich.

Es heisst **Kompilierung**. Vieleicht hast du ja die Güte, den Titel zu ändern und um was es geht. Danke.

Anubispop, 2025-07-17:
YAML‑Blöcke wie button:, number: und select: mit platform: b2500 führen zu Fehlermeldungen wie
not a valid value. generation: 2 oder
ID 'b2500_1' of type b2500::B2500ComponentV2 doesn't inherit from b2500::B2500ComponentV1.

Offensichtlich erwartet das Schema an manchen Stellen eine Generation‑1‑Komponente, obwohl generation: 2 im Haupt‑Block gesetzt ist?

tomquist, 2025-07-17:
Sollte erstmal wieder gehen. Habe den build vorerst auf 2025.6.3 gepinned und schaue mir das ganze später an.

77ccBigBore, 2025-07-20:
Habe das selbe problem, generation 2 lässt sich nicht Kompilieren.

tomquist, 2025-07-20:
Fixed via #171

closed 2025-07-20 by tomquist

tomquist, 2025-07-21:
Hat mit dem Projekt zu 99% nichts zu tun.

DerEmder, 2025-07-21:
Hat tatsächlich nichts mit dem Projekt zu tun (mit ESPHome Update auch nicht). War wohl reiner Zufall, das der Sensor ausgefallen ist, nachdem ich deine Software neu geflashed hatte. Ich vermute, die Sensoren sind naß geworden bzw. die Versiegelung undicht.

Also alles gut, war mein Versehen.