tomquist/esphome-b2500 #174: Keine Webseite mehr seit Neuinstallation

KNXNorbert, 2025-07-23:
Ich habe den ESP neu geflasht, seitdem bekomme ich keine Webseite mehr angezeigt.
Ping und MQTT funktioniert aber.

[esphome-b2500.config.json](https://github.com/user-attachments/files/21390989/esphome-b2500.config.json)

KNXNorbert, 2025-07-23:
Als Ergänzung:
Ich habe hier 2 Stück HMJ-2 mit der FW 108. Die sollten doch Version 2 Speicher sein, oder???
Ich habe jetzt mal testhalber mit Version 1 kompiliert (so hatte ich das vorher auch) damit habe ich wieder die Webseite.
Warum funktioniert das nicht mit der Version 2???

Thaui57, 2025-07-24:
@KNXNorbert ich habe dir doch auf dem Photovoltaikforum gesagt wie es geht und was du erwarten kannst von dem ESP tool. Ich denke du solltest das Thema schließen um andere nicht zu verunsichern.

closed 2025-07-24 by KNXNorbert